translationalneuromodeling/tapas #118: rDCM group analysis

kristinasabaroedin, 2021-01-22:
Hi,

I am interested in applying rDCM to resting-state data. I have a question regarding group-level analysis using the output of rDCM, which I’m unclear about. I was wondering if I could take the results from rDCM to the PEB framework to look at group differences or associations with behavioural measures? If not, how do you suggest running a group-level analysis?

Thanks,
Kristina

StefanFraessle, 2021-01-22:
Hi Kristina,

First of all, thank you very much for your interest in our rDCM toolbox and I hope it proves useful for you.

That's a good question. In principle, nothing speaks agains taking the rDCM results forward to a subsequent PEB analysis. All you need to do is to make sure that you store the information that is required for PEB (i.e., posterior means and covariances, negative free energy) in the format that the PEB routines would expect. Note that rDCM by default only stores the region-wise covariance matrices and not the full covariance matrix to save storage space. To obtain the full covariance matrix (which you would need for PEB), you need to set `options.evalCp = 1;` where options is one of the input arguments to `tapas_rdcm_estimate.m`. Let me know if you have any problems with that and I'd be happy to help. Having said this, we have not yet tried this combination of rDCM and PEB - so, I cannot guarantee that it will provide meaningful results. 

A more straightforward and simple approach for group-level analyses would be to simply take the posterior means (stored in `output.Ep`) from all subjects and run classical frequentist statistics like two-sample t-tests to inspect group differences (or the equivalent non-parametric tests if normality assumptions don't hold) or correlations / multiple regression analyses to inspect associations with behavioral measures.

I hope this helps. Let me know if you have any further questions.

All the best,
Stefan

kristinasabaroedin, 2021-01-22:
Dear Stefan,

Thank you very much for your detailed response. Much appreciated! I will consider which option is best for the analysis I'm planning to do. I will reach out if I have further issues.

Best,
Kristina

closed 2021-01-22 by kristinasabaroedin

Joyce-Z-J, 2021-07-23:
Dear Kristina,

I am also using the rDCM toolbox and am very interested in taking the rDCM results into PEB analysis to look at the group-level differences. Would you mind me asking if the combination of rDCM and PEB could work? Thank you so much!

All the best,
Zhizhao