translationalneuromodeling/tapas #119: PhysIO - reference to non-existent field errors when running tapas_physio_report_contrasts

switt4, 2021-01-26:
Hi,

I recently upgraded to the latest version of the TAPAS toolbox to take advantage of the PhysIO toolbox's new automatic censoring of bad data feature.

I am running into weird "reference to non-existent field" errors when I go to run the `tapas_physio_report_contrasts()` step.

Following the instructions on the PhysIO wiki, I am using the following command to call `tapas_physio_report_contrasts()`:

```
args = tapas_physio_report_contrasts('fileReport','run01/PhysIO_test.ps','fileSpm','run01/SPM.mat','filePhysIO','run01/physio_test.mat','fileStructural','/srv/diedrichsen/data/Pontine7T/anatomicals/p04-gradcorrect/anatomical.nii')
```

The SPM.mat file is a design-only file built using the standard SPM12 GUI and containing just the regressors in the output `multiple_regressors.txt` file, as this was what was recommended for previous versions of PhysIO.  When I run the above command, I get the following error:

```Reference to non-existent field 'swd'.

Error in tapas_physio_create_missing_physio_contrasts (line 50)
fileSpm = fullfile(SPM.swd, 'SPM.mat');

Error in tapas_physio_report_contrasts (line 149)
SPM = tapas_physio_create_missing_physio_contrasts(SPM, model,
namesPhysContrasts);
```

I can manually add the missing SPM.swd field and save the SPM.mat file back out, but when I go to re-run the command `tapas_physio_report_contrasts()`, I just get a new "Reference to non-existent field" pointing to another missing field, `xCon`.  

Clearly something has changed from previous versions as to how this SPM.mat file should be created.  Is there any documentation on how this SPM.mat file should be built?  I think I've been through all of the current documentation, and I cannot find anything other than deprecated documentation for older versions of the PhysIO toolbox.

I get this error on multiple setups using Matlab 2020a and SPM12 (local machine) and Matlab 2019b and SPM12 (server).

Any help would be appreciated.

Thanks,

Suzanne

mrikasper, 2021-01-26:
Dear Suzanne,

thank you for upgrading PhysIO and continuing to use it! I would assume that PhysIO always needed an estimated SPM model (and not just a design specification) in order to report the contrasts. It will generate the contrasts that relate to the multiple regressor parts (cardiac, respiratory etc.), but it needs, for example, the beta images to be estimated to form the contrast images via SPM.

Could you try to run it after estimating the SPM, please, and see whether the issue persists?

Thank you and all the best,
Lars

switt4, 2021-01-26:
Hi Lars,

After digging through some already estimated SPM.mat files later this afternoon, I realized that the newer version of PhysIO needed the SPM.mat with the regressors to be estimated.  The `tapas_physio_report_contrasts()` command is now running for me without error.

Thanks for replying so promptly,

Suzanne

mrikasper, 2021-01-26:
Perfect! 

I'm glad this could be resolved so quickly.

Lars

closed 2021-01-26 by mrikasper