translationalneuromodeling/tapas #120: PhysIO toolbox for simultaneous brain and spinal cord fMRI

dpfyffer, 2021-01-28:
Hi experts,

We would like to use the PhysIO toolbox to create physio regressors as nuisance variables. We acquired brain and spinal cord fMRI data simultaneously, using the sequence described in Finsterbusch et al., 2013 (https://pubmed.ncbi.nlm.nih.gov/23603283/). Within one TR of 3.6 s, we have acquired 27 contiguous slices in the brain and 18 contiguous slices in the spinal cord, in a sequential manner. Importantly, we are dealing with two distinct FOVs here (brain and spinal cord) with different inter-slice intervals. Unlike a traditional fMRI sequence, where TR=num_of_slices*inter_slice_int, in our protocol it is TR=brain_num_of_slices*brain_inter_slice_int + spine_num_of_slices*spine_inter_slice_int.

For now, we only want to create physio regressors for the brain. However, given this special sequence, we are struggling to find the correct setup. The toolbox expects a single TR, number of slices, and inter-slice interval, which does not properly reflect the nature of our slice timings. We would like to tell the toolbox that the first 27 slices were acquired at a certain inter-slice interval, and the rest of the slices with a different interval.

I hope it is clear and you have some workaround suggestions.

Thanks in advance,
Dario

mrikasper, 2021-01-29:
Dear Dario,

thank you for trying out PhysIO and the very clear description. Indeed, this does not work out of the box, but similar to a multi-band experiment, there is a simple work-around, if you know what slice you want to center your regressors on.

Typically, if you have no specific hypotheses of most important ROIs or noise centers in the brain, you would choose the center brain slice (14) as your reference slice. This means that regressor timing (and explained variance) matches best in this slice, but it shouldn't be too bad for the other slices, because 

- (a) for RETROICOR, both `cos` and `sin` regressors are included that can express a slice- (or voxel-)specific phase offset (delay)  
- (b) for HRV and RVT, you anyway often include different delayed regressors into the model.

In this case, you would determine the time (within the TR) that `idx_slice=14` is acquired as: `T_sli = (idx_slice-1)*brain_inter_slice_int`

If you then compute the relative portion within TR that this acquisition occurs, `r = T_sli / TR, say it's 0.37`, then you could just create an artificial number of slices and idx_slice that reflects the right time, e.g., `nSlices=100; onset_slice=100*r=37;, TR=3.6s` in this example.

I know this is not a perfectly flexible solution, but it should work.

If you want this to work for all brain slices with the right indices, you might get a good approximation by pretending that you have more slices acquired with brain_inter_slice_int until TR is reached, i.e.,

`nSlices_apparent = ceil(TR/brain_inter_slice_int)` and use this instead of `nSlices=27`. It's not exact (unlike the solution above), but you can use the actual slice indices and enter them even as a vector to create all slice-specific `multiple_regressors.txt` at once, e.g., `onset_slice=[1:27]`.

I hope that helps and wish you a nice weekend,
Lars

dpfyffer, 2021-02-01:
Dear Lars,

Thank you very much for your fast reply and the suggested work-around.
This was indeed very helpful!

Best regards and have a nice day,
Dario

dpfyffer, 2021-02-02:
Dear Lars,

We recently came across another question related to the one I asked above:

You nicely explained us how to create all slice-specific 'multiple_regressors.txt' files at once using the RETROICOR method.
This worked very well and now we tried to implement those txt files as multiple regressors in the 1st level analysis in SPM.
Somehow, SPM is only able to deal with one multiple regressor txt file.
If we add more than one, we get an error message during the contrast manager step:
"Error using spm_run_con, Error in contrast specification, -1 1 <- !invalid contrast"

However, we would like to implement slice-wise regressors rather than volume-wise regressors in our statistical analysis to account for e.g. multiple pulse signals recorded within one TR.

Is there any way to solve or work around this problem?

Thank you very much for your help, we really appreciate that!

Best regards,
Dario

mrikasper, 2021-02-05:
Dear Dario,

there are two aspects here:

1. I was under the impression that SPM could handle multiple `multiple_regressors.txt` files. If you look on page 356 (chapter 38.2) of the SPM 12 manual, it mentions you can enter a txt file and two mat files for different regressors. If that doesn't work, it's probably a bug to be reported to the SPM mailing list. But are you sure it is really about the two files, because it seems the GLM is estimated fine and only the contrast generation fails. Maybe you have a screenshot of the design matrix?
2. As far as I know, SPM does not support slice-wise GLMs natively. If you include all slice-wise regressors in one GLM, you will have high correlations between regressors, which might hamper the sensitivity of certain PhysIO contrasts, but more importantly reduces your degrees of freedom, i.e., sensitivity for effects of interest. 
    - The simplest workaround I see is to set up several (volume-wise) GLMs with different multiple regressors-files, centered on a specific slice each.
    - I do understand your rationale for slice-wise regression, but one would hope that the impact of heart-rate changes at the scale of a TR (a few seconds) is not so high and you could get away with volume-wise regression. So maybe you check whether different volume-wise models (e.g., centered on first, middle and last slice) make indeed a difference. But again, maybe a good question to ask about slice-wise regression on the SPM mailing list.

All the best,
Lars

dpfyffer, 2021-02-09:
Dear Lars,

Thank you once more for your feedback and help, I really appreciate it.
I did some further analysis and one can indeed add more than one `multiple_regressors.txt` file.
It really seems that the contrast generation failed due to the attempt of creating a slice-wise GLM.

As a next step, I will compare the different volume-wise models as you suggested.

Thank you very much and best regards,
Dario