translationalneuromodeling/tapas #121: Physio- running batch script, missing RVT regressors?

Ashrmis, 2021-01-29:
I am trying to include RVT regressors. So in batch script( that I just run from SPM) I set:
`matlabbatch{}.spm.tools.physio.model.rvt.delays=5`;

According to the physio documentsion PDF, FAQ #10 the order of the regressors for RVT would be:

`RVT [nDelaysRVT]`

 So since I set 5, I should get 5 RVT regressors right?

But in the physio.model.R_column_names I get the expected Retroicor cardiac and respiratory regressors, but instead of seeing 5 RVT regressors I see a single {‘RVT * RRF’} regressor. Is this an error? I have attached my full batch script here if that helps.
```
matlabbatch{1}.spm.tools.physio.save_dir = {'physio_log_match_test'};

matlabbatch{1}.spm.tools.physio.log_files.vendor = 'Siemens_Tics';

matlabbatch{1}.spm.tools.physio.log_files.cardiac = {'20201208_155532_PULS.log'};

matlabbatch{1}.spm.tools.physio.log_files.respiration = {'20201208_155532_RESP.log'};

matlabbatch{1}.spm.tools.physio.log_files.scan_timing = {'20201208_155532_AcquisitionInfo_ep2d_bold_SMS 2.0X2.0X2.0 1500 TR_PMU.log'};

matlabbatch{1}.spm.tools.physio.log_files.sampling_interval = [.005, .02]; % time between 2 rows in the Puls, Resp log files. first put cardiac then respiratory

matlabbatch{1}.spm.tools.physio.log_files.relative_start_acquisition = 0;

matlabbatch{1}.spm.tools.physio.log_files.align_scan = 'last';

matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Nslices = 44;

matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.NslicesPerBeat = 44;

matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.TR = 1.5;

matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Ndummies = 0;

matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Nscans = 180;

matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.onset_slice = 1;

matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.time_slice_to_slice = [];

matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Nprep = [];

matlabbatch{1}.spm.tools.physio.scan_timing.sync.scan_timing_log = struct([]);

matlabbatch{1}.spm.tools.physio.preproc.cardiac.modality = 'PPU';

matlabbatch{1}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.min = 0.4;

matlabbatch{1}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.file = 'initial_cpulse_kRpeakfile.mat';

matlabbatch{1}.spm.tools.physio.preproc.cardiac.posthoc_cpulse_select.off = struct([]);

matlabbatch{1}.spm.tools.physio.model.output_multiple_regressors = 'multiple_regressors.txt';

matlabbatch{1}.spm.tools.physio.model.output_physio = 'physio.mat';

matlabbatch{1}.spm.tools.physio.model.orthogonalise = 'none';

matlabbatch{1}.spm.tools.physio.model.censor_unreliable_recording_intervals = false;

matlabbatch{1}.spm.tools.physio.model.retroicor.yes.order.c = 2;

matlabbatch{1}.spm.tools.physio.model.retroicor.yes.order.r = 2;

matlabbatch{1}.spm.tools.physio.model.retroicor.yes.order.cr = 0;

matlabbatch{1}.spm.tools.physio.model.rvt.yes.delays = 5;

matlabbatch{1}.spm.tools.physio.model.hrv.no = struct([]);

matlabbatch{1}.spm.tools.physio.model.noise_rois.no = struct([]);  

matlabbatch{1}.spm.tools.physio.model.movement.no = struct([]);

matlabbatch{1}.spm.tools.physio.model.other.no = struct([]);

matlabbatch{1}.spm.tools.physio.verbose.level = 2;

matlabbatch{1}.spm.tools.physio.verbose.fig_output_file = '';

matlabbatch{1}.spm.tools.physio.verbose.use_tabs = false;

```

mrikasper, 2021-01-29:
Dear Ash,

the `delays` variable is actually the delay time (in seconds) rather than the number of regressors. It can be a vector, for example, to create 2 regressors, you could enter `[0 5]` and it would create one with no delay and one 5s shifted).

In general, all parameters are documented either in the batch editor interface or if you check the comments in `tapas_physio_new`.

All the best,
Lars

Ashrmis, 2021-02-01:
Hi Lars,
Yes you are correct, thank you for pointing me the right direction. I do get 5 {RVT*RRF} regressors now.

closed 2021-02-01 by Ashrmis