translationalneuromodeling/tapas #125: rDCM: SNR for rsfMRI simulation

zenkavi, 2021-03-10:
Hi Stefan,

I was reading your HBM paper again and was confused about how you set the SNR to [0.5 1 3] when simulating rsfMRI. Did you modify the observation noise variable defined as 

```
e    = spm_rand_mar(T,n,1/2)/4;
```
in the original `DEM_demo_induced_fMRI.m` or do you have a more clever way of doing this?

Best,

Zeynep

StefanFraessle, 2021-03-10:
Hi Zeynep,

Yes - I vary the contribution of the noise when combining the "noise-free" predicted BOLD signal time series and the observation noise. Specifically, this can be done by

```
% observation noise process
e = spm_rand_mar(T,Nr,1/2)*1/4;

% compute multiplicative factor
r = diag(std(y)/SNR);

% noisy BOLD signal time series
DCM.Y.y  = y + e*r;
```

where `y` is the noise-free BOLD signal time series, `SNR` is the signal-to-noise ratio that you're aiming for, and `e` is the observation noise. This code snippet stores the noisy BOLD signal time series with the desired SNR in `DCM.Y.y`.

Hope this helps.

All the best
Stefan

zenkavi, 2021-03-10:
Perfect, thank you!