translationalneuromodeling/tapas #130: PhysIO - denoising did not work as intended

BramBeckers, 2021-03-22:
Dear Lars,

I am currently using the PhysIO toolbox for denoising of 7T multi-echo data.
My goal is to compare denoising using phys logs to data-driven approaches, e.g. multi-echo ICA.
I therefore have a preprocessed (but uncleaned) optimally combined dataset that I want to clean using RETROICOR regressor files. I have Siemens (VB) .puls and .resp files. I used a batch script which didn't run into any trouble. I have used the multiple regressor files in two ways:

1) as regressor file input in the SPM first level GLM. When doing this, my first level results still look very noisy (lots of clusters, most of which not in grey matter)
2) as regressor to create a confound corrected time-series in the CONN toolbox. This toolbox also creates QC figures, eg. plots of connectivity value distribution. After denoising these plots should approximate a normal distribution, but nothing really seems to change (i.e. I still see a skewed distribution in most cases).

It would seem that I am not running the toolbox correctly, or at least in a way that it doesn't provide valid multiple regressor files. Could you please have a look at the input & output to see what might have gone wrong (if not the input, perhaps something related to the multi-echo nature of the data?)? 

Note that for synchronization I also tried the nominal option and setting the relative start of acquisition at 25.7s (which I know is pretty much the same for all puls and resp files), but this didn't change the outcome of the denoising.

Many thanks in advance.
Best regards,
Bram

mrikasper, 2021-03-22:
Dear Bram,

thank you for using PhysIO. 

I am currently busy with grant writing, but will probably have time to look at your question in more detail after April 1st.

In the meantime, some short pointers:

-	Which PhysIO version are you using? The latest ones have quite some improvements for processing respiratory data (>= v7.3)
-	you say your first level results look noisy, do you mean the task contrasts or the PhysIO contrasts? Since you are using CONN, it’s probably resting state data, so I assume the latter? In any case, it would be great to see some F-contrast maps (as generated with tapas_physio_report_contrasts) of the cardiac, respiratory, and interaction (maybe also HRV and RVT) regressors. 
    - Typically, cardiac RETROICOR is very robust (you probably looked at representative maps, e.g., in our toolbox paper: https://doi.org/10.1016/j.jneumeth.2016.10.019 -> your maps should look similar
-	I think other researchers (e.g., Isabel Berwian) have used the regressor files successfully in CONN. But I would say this is a rather indirect measure of the efficacy of the denoising, and would first try to make sure that the F-contrasts mentioned above look right.
-	If they don’t, timing (sync between scan and phys recordings) is the most likely culprit, that would need some deeper dive into the data
-	The second most common problem is in the preprocessing and peak detection of the cardiac data, and I do notice that peak detection of the PPU didn’t work well in the example you sent separately (stems are not at maxima in screenshot). What peak detection method did you use? Any deviation from the defaults?
[image]

-	Similarly, I also wonder what happens if you switch off the censoring module. Not sure whether it’s doing the right thing here. 

I hope that helps for now.

All the best,
Lars

BramBeckers, 2021-03-22:
Dear Lars,

Many thanks for your quick reply, this is already very helpful.

To answer your questions:
- I was using version 7.2, so I updated to the latest version (v8.0). Thanks for pointing this out! Running the new module with default settings seemed to have fixed the peak detection issue.

[image]

- My study uses a pseudo-block pharmacological design, similar to the one described here: doi 10.1038/s41598-018-19444-5 (Zhao et al Sci Rep 2018). 
The first level t-contrasts are fixed duration blocks that compare BOLD-signal vs a pre-treatment baseline. I actually mainly used CONN because it simplifies batch preprocessing and incorporates aCompCor.

- I tried to create the F-contrasts as suggested, but oddly only succeeded once. I also doubt that the output will be correct, as I need to provide the physio.mat file and only provided 1 whereas I have 2 files (from 2 sessions). If I now try to run the tapas_physio_report_contrasts script, it throws the following error:

Index exceeds array bounds.

Error in tapas_physio_overlay_contrasts (line 131)
        matlabbatch{1}.spm.stats.results.conspec.titlestr = [titleGraphicsWindow ' - ' SPM.xCon(idxContrast).name];

Error in tapas_physio_report_contrasts (line 162)
tapas_physio_overlay_contrasts( ...

Even though I used a new first level results folder.

- Finally regarding sync between scan and phys logs, this is still one of my concerns. On the positive side, PhysIO does seem to recognize the difference between log and first DICOM, which approximates 25.7s. However, since I am using (combined) multi-echo data I am not sure if this approach is sufficient. I used the first DICOM containing the second echo, which should be closest to the real TE in the optimally combined files used for fMRI analysis. There will likely be a remaining (minor) difference in timing, though I am not sure how significant this difference will really be.

Best regards,
Bram

BramBeckers, 2021-04-20:
@mrikasper Just a friendly reminder, would you have any additional thoughts on this? I have looked into the peak detection and synchronization and both appear to be fine (though with the latter the multi-echo acquisition could be problematic, as noted above). I am quite stuck with finding the issue.

Many thanks in advance.
Best regards,
Bram

mrikasper, 2021-05-25:
Dear Bram,

sorry for the delayed response. 

- Regarding `tapas_physio_report_contrasts`. Could you try to setup a GLM just with the data from one run and use it with the respective physio.mat? The evaluation per run should be sufficient to assess the denoising performance.
- The multi-echo timing between echoes should really be negligible. It's probably on the order of 30ms, which is 3% for typical heartrates of 60 bpm. So if the contrasts in the GLM above look reasonable, I would assume the sync was successful.

All the best,
Lars

closed 2022-02-20 by mrikasper