translationalneuromodeling/tapas #132: PhysIO RETROICOR Respiratory and Interaction Regressors Filled with 0s (Siemens VB)

ahartleupmc, 2021-04-02:
Hello,

I'm working on running PhysIO with Siemens .puls, .resp, .dcm files across several different tasks. For some of my test subjects everything seems to run fine, while for others the respiratory and interaction regressors of RETROICOR are completely filled with 0s. Interestingly, for those subjects some of the tasks seem to run fine, while certain tasks have this problem. Below is an example of the script file used to run one of the tasks that does not work. For reference, I am using PhysIO 4.0.0, although I did briefly try using 5.0.0 this morning without any luck.

%% Create default parameter structure with all fields
physio = tapas_physio_new();

%% Individual Parameter settings. Modify to your need and remove default settings
%physio.save_dir = {'physio_out'};
physio.log_files.vendor = 'Siemens';
physio.log_files.cardiac = {'pulse.puls'};
physio.log_files.respiration = {'resp.resp'};
physio.log_files.scan_timing = {'task.dcm'};
physio.log_files.sampling_interval = [0.0025 0.02];
physio.log_files.align_scan = 'first';
physio.scan_timing.sqpar.Nslices = 70;
physio.scan_timing.sqpar.TR = 1.5;
physio.scan_timing.sqpar.Ndummies = 0;
physio.scan_timing.sqpar.Nscans = 330;
physio.scan_timing.sqpar.time_slice_to_slice = 0.021428571;
physio.scan_timing.sqpar.onset_slice = 1;
physio.scan_timing.sync.method = 'scan_timing_log';
physio.preproc.cardiac.modality = 'PPU';
physio.preproc.cardiac.filter.include = 1;
physio.preproc.cardiac.filter.type = 'butter';
physio.preproc.cardiac.filter.passband = [0.3 9];
physio.preproc.cardiac.initial_cpulse_select.method = 'auto_matched';
physio.preproc.cardiac.initial_cpulse_select.max_heart_rate_bpm = 120;
physio.preproc.cardiac.initial_cpulse_select.file = 'initial_cpulse_kRpeakfile.mat';
physio.preproc.cardiac.initial_cpulse_select.min = 0.4;
physio.preproc.cardiac.posthoc_cpulse_select.method = 'off';
physio.preproc.cardiac.posthoc_cpulse_select.percentile = 80;
physio.preproc.cardiac.posthoc_cpulse_select.upper_thresh = 60;
physio.preproc.cardiac.posthoc_cpulse_select.lower_thresh = 60;
physio.model.output_multiple_regressors = 'multiple_regressors.txt';
physio.model.output_physio = 'physio.mat';
physio.model.retroicor.include = 1;
physio.model.retroicor.order.c = 3;
physio.model.retroicor.order.r = 4;
physio.model.retroicor.order.cr = 1;
physio.model.rvt.include = 1;
physio.model.rvt.delays = 0:5:20;
physio.model.hrv.include = 1;
physio.model.hrv.delays = 0:6:24;
physio.verbose.level = 0;
physio.verbose.process_log = cell(0, 1);
physio.verbose.fig_handles = zeros(1, 0);
physio.verbose.use_tabs = false;
physio.verbose.show_figs = false;
physio.verbose.save_figs = false;
physio.ons_secs.c_scaling = 1;
physio.ons_secs.r_scaling = 1;

%% Run physiological recording preprocessing and noise modeling
physio = tapas_physio_main_create_regressors(physio);

huppfi, 2022-05-16:
Any news on this?

mrikasper, 2022-05-16:
My apologies, @huppfi @ahartleupmc,

I was traveling to ISMRM, so things got a bit delayed.

If the respiratory and interaction regressors are zeros, it suggests that the respiratory traces might be corrupted (detaching of the respiratory bellows or similar?). Could you paste the output figures from PhysIO (`verbose.level=1`) here, please, in particular the first ones with diagnostics and raw data?

All the best,
Lars