translationalneuromodeling/tapas #136: Error message when running script

katamolnar, 2021-05-07:
Hi! I get this error message when I try to run PhysIO script with respiratory and cardiac data:
[image]
[image]

To enter the information about the data I used a guide that has worked well in the past. Here are the specifications:
[image]
[image]
[image]
[image]
[image]

I tried to find the problem but I'm not very familiar with the toolbox and so far, it’s I couldn't solve it. Do you know what the issue might be?
Thank you for your help!

mrikasper, 2021-05-25:
Dear Kata,

my apologies for the late reply, with ISMRM and the pandemic still going on. Line 88 in `tapas_physio_siemens_table2cardiac` computes 

```
endClipSamples = floor(endCropSeconds/dt);
```

Presumably one of the two variables is not set properly. It might be that `dt` has two values (for cardiac and resp. data) or that the crop is empty. Could you check these values via a breakpoint on error, please? Alternatively, you can also send me the SPM batch file and the logfiles via e-mail and I try to reproduce the error.

All the best,
Lars

mrikasper, 2021-05-28:
Dear Kata,

thank you for providing the logfiles and matlabbatch. There were two errors in the batch specification:

1. Your logfile `vendor` is `BioPac_mat`, not `Siemens_VB`. Therefore, the wrong readin function was used by the toolbox
2.  `TR` has to be specified in seconds (not milliseconds), which invoked an out-of-memory error when padding the data. Usually SI units are used in PhysIO, but if unsure, you can always check the info window in the batch editor about each parameter (see screenshot).

With these adjustments, your example runs through on my system. However, your ECG data looks irregular: no clear R-peak, there seem to be several (Q? S?) subpeaks of near-equal amplitude, making peak detection for cardiac RETROICOR hard. You can see this in the Preproc Diagnostics Plot. Can you check for other subjects whether this is a recurring issue? The respiratory data looks fine though.

[image]
[scripts_job_LK.txt](https://github.com/translationalneuromodeling/tapas/files/6562942/scripts_job_LK.txt)

[image]

All the best,
Lars

closed 2022-02-20 by mrikasper