translationalneuromodeling/tapas #138: In relation to errors when using rDCM toolbox

tyatomi0806, 2021-06-06:
Dear Stephan,

  Nice to e-meet you, my name is Taisuke Yatomi. I am a psychiatrist in the Department of Neuropsychiatry, Keio University School of Medicine, Japan. Thank you very much for sharing your TAPAS with everyone. I would like to use TAPAS rDCM toolbox to analyse resting-state functional MRI data. 

  Could you kindly teach me how to deal with the following errors when I was running its code? Thus far, I have done “tapas_rdcm_tutorial()” code and have made DCM structure data (N=15) by SPM12 (while referring “DCM for resting state fMRI” chapter in SPM12 manual) successfully. 

 Thank you for your support in advance! 

```
>> tapas_rdcm_tutorial()
Generate synthetic data

Regression dynamic causal modeling (rDCM) 

Run model inversion
Processing p0: 1/1
Processing region: 50/50
Finalize results
経過時間は 628.141700 秒です。
(This Japanese sentence means “Elapsed time is 628.141700 seconds”. )

Summary

Accuracy of model architecture recovery: 
Sensitivity: 0.918 - Specificity: 0.992
Root mean squared error (RMSE): 0.106

(# So far, I could use “tapas_rdcm_tutorial()”, and its result seems to the same as that in rDCM toolbox manual (by Taisuke Yatomi).)

>> [DCM_XXX_setop]=tapas_rdcm_set_options('DCM_XXX.mat', [], 's')
この型の変数ではドット インデックスはサポートされていません。
(#Comment: This Japanese sentence means that “dot index is not supported with the variable of this type”.)

エラー（=error）: tapas_rdcm_set_options (line 109)
r_dt = options.y_dt/DCM.U.dt ;
 
>> load('DCM_XXX.mat')
>> [output]=tapas_rdcm_estimate('DCM_XXX.mat', 's', [], 1)

Regression dynamic causal modeling (rDCM)

存在しないフィールド 'Y' を参照しています。
(#Comment: This Japanese sentence means that “the code is loading the field ‘Y’ that does not exist”. )

エラー(=error): tapas_rdcm_estimate (line 74)
    DCM.U.u     = zeros(size(DCM.Y.y,1)*16, 1);
```

Sincerely, 
Taisuke

Taisuke Yatomi, M.D.

StefanFraessle, 2021-06-07:
Hi Taisuke,

Thank you very much for your interest in rDCM - I hope it proves useful for your project. 

Thank you for your question. From what I can see, the error occurs because of the way how you provide the DCM structure to `tapas_rdcm_estimate()`. Specifically, the function requires you to provide the DCM structure and not the filename. Hence, it should work when running:

`load('DCM_XXX.mat')`
`[output,options] = tapas_rdcm_estimate(DCM, 'r', [], 1)`;

Two things are important to note:
- First, you should use the option 'r' instead of 's' when calling `tapas_rdcm_estimate()`. This is the correct option when analyzing empirical data.
- Second, I would suggest to set up the model using the function `tapas_rdcm_model_specification()`. This function will ensure that the DCM structure is correctly specified - if you set-up the DCM within SPM, there might be compatibility issues which lead to errors in the code. Setting up a resting-state fMRI analysis with `tapas_rdcm_model_specification()` is quite simple. All you need to do is provide the BOLD signal time series to the function. This is done as follows: `[ DCM ] = tapas rdcm model specification(Y, [ ], [ ])` where Y is the data in the typical SPM format - that means, containing the field `Y.y` (time series), `Y.dt` (repetition time), and `Y.names` (region names). You can also consult the rDCM Manual and the help of the function `tapas_rdcm_model_specification()` for more information.

Hope this is helpful.

With very best wishes,
Stefan

tyatomi0806, 2021-06-27:
Dear Stefan, 

 Thank you for your immediate and extremely useful advice for my questions, and I was also very sorry for my error on your name in the former github chat. On the other hand, there are two more questions that I would like to ask you after conducting data analysis with rDCM toolbox. According to your advice, I have managed to conduct "tapas_rdcm_estimate" command, but ①could you please teach me how to interpret its result after doing another "tapas_rdcm_visualize" *command, especially in relation to contrasts between region(from) and region(to) in the estimated segment (is there any means to assess which region is the most afferent, or which region cause more efferent signals than other regions in the result of tapas_rdcm_visualize or tapas_rdcm_estimate itself?) and/or Pp binary indicatior (is it all right if all sections of Pp binary indicator are yellow in the following example?) in the following example due to the result " tapas_rdcm_estimate(DCM, 'r', [], 1)" from Default Mode Network regions (PCC, mPFC, LIPC, and RIPC, from the data http://www.fil.ion.ucl.ac.uk/spm/data/spDCM/)?

(* After the code of "caxis" function in "tapas_rdcm_visualize" many parts, I inserted 
```
xticklabels({'PCC', 'mPFC', 'LIPC', 'RIPC'})
yticklabels({'PCC', 'mPFC', 'LIPC', 'RIPC'})
```
)  
[tapas_rdcm_visualize_result_of_DMN_DCM(Yatomi).pdf](https://github.com/translationalneuromodeling/tapas/files/6721715/tapas_rdcm_visualize_result_of_DMN_DCM.Yatomi.pdf)

 Secondly, ②would you mind teaching me how you made "Volume of Interest regions" of networks "SAN", "DAN", "DMN", and "CEN" using SPM or other softwares or codes when you wrote your paper "Regression dynamic causal modeling for resting-state fMRI(Hum Brain Mapp. 2021;42:2159-2180)"? I have already read the paper "Decoding Subject-Driven Cognitive States with Whole-Brain Connectivity Patterns by Shirer et al., 2012" and downloaded the anterior and posterior salience-network nii. file, dorsal and ventral default mode network nii. file, etc in the indicated website (The FIND lab's HP, http://findlab.stanford.edu/functional_ROIs.html), but I could not be able to utilize them to extract those networks' volumes of interest as I mentioned above having their BOLD time series from my rsfMRI data set using SPM12 (I have mainly used its batch editor and the command "Volume of Interest", but I could not extract network volumes). 

Best Regards, 
Taisuke

Taisuke Yatomi, M.D.

closed 2021-06-28 by tyatomi0806