translationalneuromodeling/tapas #139: Computational modeling using HGF

emilia-kn, 2021-06-18:
Hello,

I am trying to use HGF Toolbox to model prediction errors in a probabilistic reversal learning task. I read the Archives which was very helpful, but still I have some doubts. Hopefully you can help me to clarify them.

In our task cue A was followed by a reward 70% of occasions and a neutral feedback 30% of occasions, while for cue B the reinforcement schedule was reversed (70% of occasions no reward/neutral feedback, 30% of occasions reward). As far as I understand this should be coded as:
cue A -> reward =>input 1
cue B -> no reward/neutral feedback => input 1
cue A ->  no reward/neutral feedback => input 0
cue B -> reward => input 0

Because this was reversal learning, after 5 consecutive correct responses the reinforcement schedule was reversed (cue B was followed by reward and cue A by neutral feedback 70% of times). In such case, should the coding be also changed (cue A -> reward and cue B -> no reward/neutral feedback coded as 0; cue A ->  no reward/neutral feedback and cue B -> reward coded as 1) or should it stay as it was? And if I leave the codinig without change, will the interpretation of expected values be "What is the probability of obtaining reward given that cue A appeared?"

Secondly, subjects who participated in the study knew that one cue is followed by a reward more often than the other and that the probabilities sum up to 1 (so they knew that if A is followed by a reward with probability X, B is followed by reward with probability 1 - X). Does this knowledge affect the way the input is coded?

Finally, does the way of coding the input and the way the prediction errors (PEs) are calculated have any impact on modeling fMIR data using GLM approach? So for example, can I model the trials dividing them based on the prediction error value (which should be negative when the input was 0 and positive when the input was 1) as well as model all the trials together regardless of the PE value? Or for example if after reversal the coding is not changed, then should I put in GLM cue A and cue B as separate regressors? Wouldn't be putting all the cues that predict a reward with higher probability together as one regressor more informative way of designing GLM?

Hope my questions are clear and I'm sorry if they are naive, but I'm just beginning in this field.
Thank you very much in advance,
Emilia

chmathys, 2021-06-30:
Dear Emilia,

Some responses are below.

Best wishes,
Christoph

> Hello,
> I am trying to use HGF Toolbox to model prediction errors in a probabilistic reversal learning task. I read the Archives which was very helpful, but still I have some doubts. Hopefully you [...]
This makes sense.
> Because this was reversal learning, after 5 consecutive correct responses the reinforcement schedule was reversed (cue B was followed by reward and cue A by neutral feedback 70% of times). In such [...]
It should stay as it was.
>  And if I leave the codinig without change, will the interpretation of expected values be "What is the probability of obtaining reward given that cue A appeared?"
Yes. Equivalently, the interpretation is also “probability of obtaining *no* reward given cue B”.
> Secondly, subjects who participated in the study knew that one cue is followed by a reward more often than the other and that the probabilities sum up to 1 (so they knew that if A is followed by a [...]
It means the way you’ve coded it corresponds to the instructions of the participants, so that’s good.
> Finally, does the way of coding the input and the way the prediction errors (PEs) are calculated have any impact on modeling fMIR data using GLM approach? So for example, can I model the trials [...]
Since your choice of which outcome to code as 1 and which as 0 is arbitrary, the sign of the prediction error at the first level is not interpretable. Because of this, the absolute value of the prediction error was used as a GLM regressor in similar cases.
>  Or for example if after reversal the coding is not changed, then should I put in GLM cue A and cue B as separate regressors? Wouldn't be putting all the cues that predict a reward with higher probability together as one regressor more informative way of designing GLM?
Which regressors you want to include depends on your research question. Given that, it should become clear which ones to include.

emilia-kn, 2021-07-05:
Dear Christoph,

thank you for your answer, it's really helpful. I have 2 short follow up questions:
1. You wrote "the sign of the prediction error at the first level is not interpretable". What do you specifically mean by the "first level"? Could you give me some advice how can I compute prediction errors so that I can use them in fMRI data analysis to look for the brain correlates of negative and positive prediction errors? We used RW, Sutton and HGF models.
2. Since information about cues is always complementary, we thought that considering these two events (cue A and cue B) separately does not add anything and considered a new coding, where reward = 1, no reward = 0. It is not about calculating a new model for this new coding, but presenting the results of the model based on coding I described in the previous message (cue A -> reward and cue B -> no reward coded as 1; cue A -> no reward and cue B -> reward coded as 0) in terms of new coding.
Prediction errors are expressed as s - P( 1 | q), where s - state coded as 1 or 0 and P( 1 | q ) is a probability of state 1 providing cue q. P( 1 | q) for q = "cue A" are the calculated expected values:

P("cue A -> reward" or "cue B -> no reward" | "cue A") = P("cue A -> reward" | "cue A") = P("cue A -> reward") = P("reward" | "cue A")

To obtain P(1 | q) for q = "cue B" in the convention where the states are defined by new coding, we changed the expected values x to 1 - x. In other words such values are probabilities of getting a reward providing cue B:

P("cue A -> reward" or "cue B -> no reward" | "cue B") = P("cue A -> reward" | "cue B") = P("cue A -> reward") = 1 - P("cue B -> reward") = 1 - P("reward" | "cue B")

This I believe changes the interpretation of expected value to "What is the probability of obtaining reward given that I see what I see?" rather than "What is the probability of obtaining reward given that cue A appeared? or “What is the probability of obtaining *no* reward given cue B?” 
What do you think about such solution?

Thank you again,
Emilia

chmathys, 2021-09-19:
Dear Emilia - some quick remarks below.

> Dear Christoph,
> thank you for your answer, it's really helpful. I have 2 short follow up questions:
>  1. You wrote "the sign of the prediction error at the first level is not interpretable". What do [...]
I mean delta_1, which sits at the first level and drives the updates at the second level. You can find these in the first column of the est.traj.da matrix. Since the assignment of negative and positive is arbitrary in your case, it makes more sense to use the absolute value of delta_1 as a regressor.
>  2. Since information about cues is always complementary, we thought that considering these two events (cue A and cue B) separately does not add anything and considered a new coding, where reward = 1, no reward = 0.
You can do that if you simply want to model learning about the probability of reward, independent of any cues.
> It is not about calculating a new model for this new coding, but presenting the results of the model based on coding I described in the previous message (cue A -> reward and cue B -> no reward coded [...]
I’m not sure I follow your reasoning entirely, but if you’ve thought this through and are convinced it’s a good solution, then by all means go with it.