translationalneuromodeling/tapas #140: rDCM sparse

Joyce-Z-J, 2021-06-22:
Dear Stefan,

Sorry to bother you again but would you mind me asking a question about the 'tapas_rdcm_sparse' function?

When I was running the code, an error message showed up: unrecognized field name "X0" (see attached screenshot). Could you please tell me which step generates the variable ‘DCM.U.X0'? (I specified the model using scripts before so I assume that the problem is not due to incompatibility).

Thank you so much for your time and help! :)

All the best,
Joyce

[image]

StefanFraessle, 2021-06-23:
Dear Joyce,

Thank you for your question. 

In brief, `args.verbose` is set in the main function `tapas_rdcm_estimate.m` just before `tapas_rdcm_sparse.m` is called (line 140 in `tapas_rdcm_estimate.m`). 

In general, I would recommend you **do not** call the inversion functions `tapas_rdcm_ridge.m` and `tapas_rdcm_sparse.m` directly, but instead use `tapas_rdcm_estimate.m` to perform model inversion. `tapas_rdcm_estimate.m` is the main (wrapper) function of rDCM and will make sure that all of the relevant fields, arguments and options are set adequately in order for rDCM to work properly. This is not the case if you call the inversion functions directly.

Hope this helps.

All the best,
Stefan

Joyce-Z-J, 2021-06-23:
Dear Stefan,

Thank you so much for your timely reply! You do help me a lot :)

Would you mind me asking if I could perform 'tapas_rdcm_model_specification' directly to 'tapas_rdcm_estimate' or 'tapas_rdcm_model_specification' to 'tapas_rdcm_set_options', 'tapas_rdcm_create_regressors' then 'tapas_rdcm_estimate'? 

And can I ask another question regarding the result visualisation? Which input should I write in 'plot_regions' and 'plot_mode'? If there are six regions in the model structure, should the 'plot_regions' be set as [1 6]? And is 1 in 'plot_mode' for plotting power spectral density of temporal derivative and 2 for BOLD signal time series?

Thank you so much for your help!

All the best,
Joyce

StefanFraessle, 2021-06-24:
Dear Joyce,

You would first use `tapas_rdcm_model_specification.m` to set up the DCM and then directly use `tapas_rdcm_estimate.m` to perform the model inversion.

`plot_regions` refers to the regions for which you want to plot the predicted and true signal - hence, you can set these numbers to whatever you want. For instance, if you only want to inspect the signal of region 1, then you enter `1`. If you want to plot the signal of all six regions, you enter `1:6`.

`plot_mode` is exactly what you describe and distinguishes between power spectral densities of the temporal derivative and the BOLD signal time series. The default is to inspect the power spectral densities because that is what rDCM fits during model inversion. For more details, see another post that contains some information on this (#129).

Hope this helps.

With very best wishes,
Stefan

closed 2021-06-28 by Joyce-Z-J