translationalneuromodeling/tapas #143: rDCM group level analysis

Joyce-Z-J, 2021-06-26:
Dear Stefan,

Sorry to bother you again but would you mind me asking if I can perform rDCM group level analysis using Bayesian Model Selection (BMS)? My understanding (not sure if it is correct) is that: BMS is generally used in selecting the best fit model from all the alternatives, but since rDCM specifies only one model (the full model), BMS can hardly be used for rDCM group level analysis. (Please correct me if my idea is wrong).

If BMS cannot be used for rDCM group level analysis, would you mind suggesting other methods regarding this issue?

Thank you so much!

All the best,
Joyce

StefanFraessle, 2021-06-28:
Dear Joyce,

Yes, you can perform Bayesian Model Selection with rDCM. Just like for classical DCMs, model inversion yields both posterior estimates of the model parameters as well as an estimate of the negative free energy. The negative free energy is an approximation to the log model evidence and serves as a principled metric for model comparison.

Hence, once you have specified your alternative hypotheses (i.e., models) and have fitted all these models to the data, you can use BMS to select the winning model. 

For an example, see [Frässle et al. (2021)](https://www.sciencedirect.com/science/article/pii/S1053811920309769?via%3Dihub).

Hope this helps.

With very best wishes,
Stefan

StefanFraessle, 2021-06-28:
p.s.: Would you mind closing the issues that are still open and have been answered :) Only you can do this and it would help us keep the issues feed uncluttered. Thank you very much.

closed 2021-08-03 by Joyce-Z-J