translationalneuromodeling/tapas #146: Running PhysIO Toolbox

sbkeo, 2021-07-12:
Hi, I am fairly new to coding which may be why I am having some difficulty using the PhysIO Toolbox but I also can't seem to find instructions anywhere on how to load the fMRI data as well as the physio data and get it all up and running. Sorry if this is a easy question but any help would be appreciated.

mrikasper, 2021-07-12:
Dear @sbkeo,

great you are trying out PhysIO! It's designed to be accessible for coding novices as well.

1. A good starting point is still our [PhysIO publication](https://doi.org/10.1016/j.jneumeth.2016.10.019), if you want to understand the workflow (esp. Figs. 1 and 2). 
2. Also, if you are using SPM and have PhysIO installed, the Batch Editor has a lot of online help about the different parameters
3. For installation and a quick start, have a look in the [README.md](https://github.com/translationalneuromodeling/tapas/blob/master/PhysIO/README.md) and Wiki, e.g., the [Quickstart page](https://gitlab.ethz.ch/physio/physio-doc/-/wikis/QUICKSTART)

Note that in, general, the physiological noise modeling (synchronization to fMRI data, preprocessing of physiological traces, creation of regressors for later GLM analysis) is done within PhysIO. The actual _correction_, i.e., removing or co-modeling the physiological noise, is performed by the fMRI analysis package of your choice (e.g., SPM, or FSL), and PhysIO provides the `multiple_regressors.txt` file to input them as nuisance regressors. 

I hope that helps,
Lars

sbkeo, 2021-07-13:
I'm sorry, I'm still having difficulty. I use the Biopac system and when I transfer the files, they're all .acq files. Do I need to convert them to .mat or .txt files and each acquisition file also includes 3 columns of data (trigger, cardiac, resp) do I need to make each its own file so it can be properly loaded? Thank you.

mrikasper, 2021-07-13:
Dear @sbkeo,

typically, there is an option to export the Biopac files with their software into "Matlab" files `.mat`. These ones could be read in by PhysIO choosing `BioPac_mat` as `logfiles.vendor`.

I hope that helps.

All the best,
Lars

closed 2022-02-20 by mrikasper