translationalneuromodeling/tapas #149: Siemens CMRR 4-electrodes ECG data and physio regressors generation (PhysIO)

jrasero, 2021-07-20:
Hi,

How does the PhysIO toolbox handle the Siemens CMRR multiband ECG data during the generation of the physio regressors? This data has 4 signal columns coming from 4-electrodes. Is a set of regressors for each electrode generated, for a combination or maybe for the average signal? Do you know if one of these 4-electrodes is a ground electrode?

Sorry if the questions are a bit naive. Also, I haven't seem them answered in the issues open/closed in the repo, nor in the paper of the toolbox, but my apologies in advance if I am wrong.

Many thanks!

mrikasper, 2021-07-20:
Dear Javier,

thank you for your interest in PhysIO! Per default,  PhysIO selects the `v1` column. 

- There is a parameter to change this in `tapas_physio_read_physlogfiles_siemens_tics` 
    - see the header of that file, something like `tapas_physio_read_physlogfiles_siemens_tics(...
    log_files, cardiac_modality, verbose, 'ecgChannel', 'v2'))`
- it's not a UI parameter at the moment, so you would have to change it in the file directly, or adapt the call in the parent functions `tapas_physio_read_physlogfiles` and `tapas_physio_main_create_regressors` to make it flexible. 
- if you want to dig deeper, the implementation of the read-in is in a function called `tapas_physio_split_data_per_channel_siemens_tics`

Please let me know whether this functionality of channel selection is of general interest. If that's the case, we can add a feature request for a user parameter for that in a future release, or you could implement it yourself and we merge it back into our code base as a pull request.

I hope this helps!

All the best,
Lars

jrasero, 2021-07-21:
Thanks for such a quick response, Lars!!

Digging into the functions that you pointing me, specifically  `tapas_physio_read_physlogfiles_siemens_tics` and playing myself with it, I have followup question on the 'ecgChannel'  argument that you may pass to this function. I think I know the answer, because It's in the documentation of this function, but just to make sure:

- Assuming again the Siemens 4-electrode multiband ECG output, am I correct saying that setting ''ecgChannel to 'c1', 'c2', 'c3' or 'c4' will output one of these ECG columns signal, whereas 'v1' would give you the difference between two channels, in this case 'c1' and 'c2'?

Do you know if there's any preference in using one type of output or another (single channel vs difference) when trying to best accounting for physiological ECG noise in the subsequent GLM first-level models?

Thanks a lot for your help and congratulations for such a wonderful toolbox!

Best,
Javier

mrikasper, 2021-07-21:
Dear Javier,

indeed, you are right - and have spotted an inconsistency in the documentation ;-) 

- So `c1`, `c2` etc. will give you the respective channel output, but `v12` would give you the difference between the two. 
- Should work with any two channel indices, e.g., `v34` to get the difference between channel 3 and 4.
-  `v1` is just a short-hand for `v12`. Similarly, `mean` is short for mean of `v1` and "`v2`" (i.e., `v12` and `v34`)

All the best,
Lars

jrasero, 2021-07-21:
Great, thanks!!

Any hint which one is preferable or more recommended (either a single channel, a difference or the mean) to be used to compute the physiological regressors? Or is this based on the researcher's experience? 

Thanks!

mrikasper, 2021-07-21:
I think the default `v12` is probably a good start. Using voltage differences makes sense because it cancels out gradient noise, but it also depends on whether all electrodes were placed correctly on the subject. 

If you set a higher verbosity level in PhysIO (`verbose.level=3`), I think you should see a debugging plot with all the raw data channels and the selected one (e.g., combined mean or difference). There is probably no way around checking this for every subject, as human anatomy and electrode placement have a huge impact.

All the best,
Lars

jrasero, 2021-07-21:
Lars,

Thanks a lot for such a helpful reply. It makes totally sense what you've just said. I will start as a default with `v12` and check this option increasing the verbosity level as you recommend.

I am going now to close this comment, as I have no further questions.

Again, let me congratulate you guys for such a brilliant work to develop this amazing toolbox.

Cheers,
Javier

closed 2021-07-21 by jrasero