translationalneuromodeling/tapas #150: PhysIO - Error thrown with BIDS format phys data

dariusvalevicius, 2021-07-20:
Hi,

I'm trying out PhysIO on some data accompanied by preprocessed physiological data in BIDS (.tsv) format. I'm getting an error about NaN values in the resp regressors. I've tried configuring the parameters a few different ways (splitting the .tsv into two plain text files and using the 'Custom' vendor) but no dice so far. The phys data is preprocessed to match the fMRI run and so the length (24500 rows at 0.02 Hz) exactly matches the fMRI length (196 * 2.5 TR = 490s).

Any help would be appreciated!

Thanks,
Darius

MATLAB output:
>     maxscan (incl. dummies) = 196 
>     tmin (1st scan start (1st dummy))=   0.00 s
>     tmin (1st scan start (after dummies))=   0.00 s
>     tmax = 489.92 s 
>     mean TR =   2.50 s
> 
> Warning: Guessed additional cardiac pulse at time series start for phase estimation 
> > In tapas_physio_log (line 58)
> In tapas_physio_get_cardiac_phase (line 49)
> In tapas_physio_create_retroicor_regressors (line 89)
> In tapas_physio_main_create_regressors (line 294)
> In physio_cli_fmri (line 109) 
> Note: Guessed 1 additional cardiac pulse(s) at time series end for phase estimation
> Error using tapas_physio_get_respiratory_phase (line 62)
> Nan values in the respiratory regressors
> 
> Error in tapas_physio_create_retroicor_regressors (line 113)
>             r_phase = tapas_physio_get_respiratory_phase(fr,rsampint, 0);
> 
> Error in tapas_physio_main_create_regressors (line 294)
>                 tapas_physio_create_retroicor_regressors(ons_secs, sqpar, ...
> 
> Error in physio_cli_fmri (line 109)
> physio = tapas_physio_main_create_regressors(physio);

mrikasper, 2021-07-20:
Dear Darius,

thank you for trying out PhysIO! BIDS should be read in alright, I don't think you would have to convert it. Could you give me some more details on what happens exactly, e.g.:

- the version of PhysIO (`tapas_physio_version`) you use
- any output figures from PhysIO before the error (esp. the Read-in raw time series with the breathing time course, and the Preproc: Diagnostics plot)
- are there any NaNs you can spot in the BIDS file?

Thank you for your feedback,
Lars

dariusvalevicius, 2021-07-22:
Hi Lars,

- I'm running physio version R2020a-v7.3.0 (on MATLAB 2020b)

- Here's the raw timeseries plot and the preproc diagnostic plot: Resp data seems to be completely missing
[image]
[image]

- No NaNs in the bids file. Here's a look inside:
[image]

The first column is resp and the second is cardiac, I believe. Maybe it's missing a time var?

Thanks,
Darius

mrikasper, 2021-07-22:
Dear Darius,

indeed, it looks like the respiratory curve is missing, but it also looks as if the cardiac raw trace has a very long cycle length (3 seconds).

What does the accompanying `.json` header file look like? If the respiratory trace is indeed in the first column (which is unusual), then the header should reflect that, see [BIDS Physio spec](https://bids-specification.readthedocs.io/en/stable/04-modality-specific-files/06-physiological-and-other-continuous-recordings.html)
```
{
   "SamplingFrequency": 100.0,
   "StartTime": -22.345,
   "Columns": ["cardiac", "respiratory", "trigger"],
   "cardiac": {
       "Units": "mV"
   }
}
```

I don't think that a time column is missing, if you specified header `StartTime` and `SamplingFrequency` correctly.

Another guess: Did you specify the same BIDS file for both `log_files.respiration` and `log_files.cardiac`? If one is left empty, PhysIO assumes you don't want to compute this modality.

All the best,
Lars

dariusvalevicius, 2021-07-24:
Hi Lars,

Here is the header:

> {
> "SamplingFrequency": 50, 
> "StartTime": 0, 
> "Columns": ["cardiac", "respiratory"]
> }

The column order does seem to mismatch with the content of the file. I tried swapping around the contents of the .tsv to match it, but ran into the same error and lack of resp read-in.

> Another guess: Did you specify the same BIDS file for both log_files.respiration and log_files.cardiac? If one is left empty, PhysIO assumes you don't want to compute this modality.

Both parameters are set to the .tsv. It may be some problem with the dataset... It's one I pulled from openneuro.org. I'll try another (they all seem to use BIDS format for physio logfiles) and report back.

Thanks,
Darius