translationalneuromodeling/tapas #152: HGF Trajectory Issues

jgaretti, 2021-07-22:
Hi all,

I've been working with the HGF to model behavioral data for a reversal learning task with both social and non-social components. When setting up the priors for each new subject, I noticed that they were always the same across subjects for new datasets. Additionally, I've noticed that the overall shapes of the trajectories are almost only dependent on the input cues (red asterisks, see attached), and not the participant responses (orange dots). I had originally believed that the priors should be individualized per subject, and the trajectory shapes should be mainly based off the responses, while the ground truth is based off the input cues. I'm just looking for some conceptual clarification, am I mistaken about this?

Thank you very much,
Jake

[image]

adonda10, 2021-07-22:
Hi Jake, 

I am sure Chris Mathys will give more expert advise, but I can make a suggestion in the meantime: Each participant should have the same priors, and then when the model is fitted to the data you will obtain individual posterior estimates on those parameters. The posterior estimates (mean, var) will differ across participants, but the priors should be the same. That that is a requirement to make sure you are testing the "same model" in every person. Hope this helps.

Antonia

jgaretti, 2021-07-22:
Hi Antonia,

Thank you! So when I set up the inversion I've been using an observation model without parameters in order to get starting values for the perceptual model (sigma, omega and phi are the main ones that change, mu remains constant) for running the full HGF with an observation model that includes zeta. Should those starting values all be the same across participants?

Thank you,
Jake

adonda10, 2021-07-22:
Hi Jake, 

One could always choose the priors of parameters and initial values based on previous studies. However, if nobody has used your model and reported those values, then, yes:

In the HGF toolbox scripts, they recommend that we first use the perceptual model only on the series of observed inputs (no responses), to "determine the Bayes optimal perceptual parameters".  
I understand from your text that this is what you did.
The priors inside the HGF perceptual model config should be the same for all participants, but then you will get of course different posterior estimates on these parameters for each individual.

I agree that your question is important: what to do then?  Which prior parameter values should we choose now for the perceptual + observation model?

I'd love to hear Chris Mathys' answer on this, but what I'd do is to look at the distribution of posterior parameter values you obtained after running the perceptual model only (distribution across your N participants). Then I'd choose representative values for the total sample and use them as the new prior values for the perceptual + observational model.

But again, this is just my opinion and it would be great to hear what Chris and others recommend.

Best
Antonia

jgaretti, 2021-07-23:
I completely agree that those posterior estimates when running the perceptual model only should be different across participants, which is why we are concerned that all of these estimates are constant no matter the subject. Based on that image above in the first comment, my initial thoughts are that the HGF I've been running is based too much off of the input cues, which is the only thing that is constant and unchanging between participants. Apologies if I've caused any confusion about what the issue is.

- Jake

chmathys, 2021-07-26:
Hi Jake,

Everything Antonia has said is correct. Your estimates should differ between participants (insofar as their reposnses differ). Which version of the toolbox are you using? And are you using any external code in addition to what’s included in the toolbox?

Best wishes,
Christoph

jgaretti, 2021-07-27:
Hi Cristoph,

Thank you so much for your reply! I'm using a bit of code from a paper from Lara Henco beginning with "[Aberrant computational mechanisms of social learning](https://journals.plos.org/ploscompbiol/article?id=10.1371/journal.pcbi.1008162)", which I believe essentially just compiles the existing tapas modeling scripts, which I have updated with the newest versions. However, I'm having a little trouble understanding conceptually certain parts of the results. In the image below, how would you interpret the stretch in the lowest trajectory from about trial 230-260 (which I circled in green), where the trajectory probability increases, reflecting the input cues changing, even though participant behavior seems to reflect that they believe for the majority of this section that the advice would be incorrect. Some clarification here would be much appreciated.

Best,
Jake

[image]

chmathys, 2021-08-03:
Hi Jake,

You’re right that the model doesn’t seem to capture this participant’s behaviour very well in the gaze domain. One reason for this may be that the gaze weight is low, ie decisions are determined mostly by card beliefs, so predictions wouldn’t be much improved by more appropriate gaze beliefs.

Another explanation could be that this participant's gaze-domain behaviour can be characterized as a continuous descent from trusting the gaze almost completely to distrusting it almost completely, irrespective of outcomes. One of the fundamental assumptions of the model is that beliefs follow the evidence, ie the sign of the update is the same as that of the prediction error. That’s the case for most learning models I’m aware of. They only differ in how they capture the size of the update. So it’s hard for such models to explain behaviour that seems to violate this fundamental assumption. Nonetheless, a better initial value for mu2_gaze might help - see right below.

In card space, this participant seems to have a bias toward outcome 1. While they don’t seem entirely unresponsive to outcomes, the learning rate also seems generally a bit too high in this domain. One thing you could try here is to estimate the initial value of mu2_card to capture this bias. The same could help for mu2_gaze because there initially seems to be a high trust in gaze.

I hope this helps.

Best wishes,
Christoph

closed 2021-09-09 by chmathys