translationalneuromodeling/tapas #153: Example fMRI data (PhysIO)

dariusvalevicius, 2021-08-02:
Hi,

Are the fMRI data corresponding to the example physlog data for PhysIO openly available? The link provided in the Kasper et al. paper (https://www.tnu.ethz.ch/en/software/tapas/data.html) appears broken.

Thanks,
Darius

mrikasper, 2021-08-04:
Dear Darius,

thank you for asking. Yes, indeed, the example data is publicly available and can be downloaded by the following procedure in the Matlab command window:

1. `tapas_init`
2. `tapas_dowload_example_data`

It will create a subfolder `examples/<tapas-version>/PhysIO` with a folder structure reflecting the different vendor formats. The adapted example `.m/.mat` files for the SPM job manager and `.m` pure Matlab files are right in the respective folder with the example data.

In general, the best starting point to find up-to-date information on the toolbox is the [PhysIO README](https://github.com/translationalneuromodeling/tapas/blob/master/PhysIO/README.md), which I curate for every new release. The paper is still a great resource, but more for the concepts and algorithms used in the toolbox. 

All the best,
Lars

P.S: I just checked: A variant of the above procedure is outlined under "Getting Started" in the README, but it might be that the `tapas_init` is crucial for the download to run error-free.

P.P.S.: If you still want to use the manuscript links, the ethz-links in the manuscript have to be updated following a change in the university's content management system, just remove the `.html`: https://www.tnu.ethz.ch/en/software/tapas/data

dariusvalevicius, 2021-08-04:
Thanks Lars,

I was wondering specifically about the BOLD data associated with the physiological recordings present in the example package. I notice in the corrected link you sent me that the wiki links to the HCP database for one of the examples. Do you know if any of the other datasets are accessible?

Best,
Darius

mrikasper, 2021-08-04:
Dear Darius,

I see now! As described in the [wiki](https://gitlab.ethz.ch/physio/physio-doc/-/wikis/EXAMPLES) you mentioned, a lot of collaborators have provided the logfile data for the different vendors. You would have to contact them directly if you are interested in the scan data.

Are you interested in such data for a specific purpose? 

If it is about understanding how to use the regressors in practice, I always wanted to provide an example dataset including imaging data to have a complete PhysIO walk-through tutorial including GLM performance assessment, but wasn't sure which public dataset to start from. Any suggestions?

All the best,
Lars

dariusvalevicius, 2021-08-04:
Hi Lars,

I'm writing a shell script in MATLAB for taking parameters, configuring and running PhysIO, and applying a correction to the fMRI run - but I'm a little short on data to test it on, hence looking for different sources. Currently I've only successfully run it on some in-house data, but it looks like I'll have to download one of the big HCP datasets next.

I'm not sure what public data would be most appropriate for a tutorial. As far as I can tell, HCP data can't be downloaded piecemeal, making it unwieldy to use. The only other thing I've found is the BIDS format data from openneuro that I wrote about in another issue, which can be downloaded more efficiently, but it has that problem with read-in in PhysIO and I'm not sure why. (Many of the datasets on openneuro appear to be broken - I actually can't find the one I used now...)

Best,
Darius

mrikasper, 2022-02-20:
Dear Darius,

My apologies for the late reply.

In case you are still looking for example datasets for PhysIO that contain scan data, I can recommend this one: 

https://dataverse.nl/dataverse/rt-me-fmri

Stephan Heunis (@jsheunis) has worked with PhysIO, so I am sure the data provided by his group is compatible with your requirements.

All the best,
Lars