translationalneuromodeling/tapas #156: Cardiac peak detection - Missing half of the peaks

cmurillo1291, 2021-09-09:
[image]

mrikasper, 2021-09-10:
Dear Carlos,

thank you for using PhysIO, I am happy you find it user-friendly! PhysIO uses the maximum expected heart rate (in beats per minute) as a prior for the heartbeat detection, which is the last one in your list, i.e., `max_heart_rate_bpm`. You could try to set it to 120 instead of 90.

The screenshot of your output is a bit blurry - I cannot read the x axis properly - but this might solve the issue, it looks like about 16 heartbeats in 10 seconds, which corresponds to 96 bpm. The signal itself looks clean enough for peak detection.

Let me know if this resolves the issue!

All the best,
Lars

cmurillo1291, 2021-09-10:
Dear Lars, 

Yes, it works!
Thank you very much for your quick response.

mrikasper, 2021-09-10:
Great to hear! Good luck with the analysis.

closed 2021-09-10 by mrikasper