translationalneuromodeling/tapas #157: Optimal parameter estimation for observational data

fahd7yazin, 2021-09-18:
Dear all,

I’m new to this toolbox and am slowly picking up the basics off the (precise) documentation and from Dr Mathys’ works. I would greatly appreciate some direction on my current research. 

My situation: I have continuous data (bounded range) of a variable from a group (ex control grp). I’m interested in using HGF on this to estimate some parameters, which would be compared to another different variable from another group (say clinical grp). Essentially, I’m not wanting to model subjectwise behaviour, rather only estimate the model parameters to be used as a comparative measure for my main group.

Which route would suit my need optimally?

1)	Estimate a 2 or 3 level HGF with empty responses ([]) using Bayes_optimal_config, and compare models with different prior parameter values and adjudicate with LME. 
2)	Simulate agent responses in different models using simModel, and recover the parameters with fitModel with the same priors used for the simulation (entered in the hgf_config file)

If there’s a better, third way please let me know.

Regards

chmathys, 2021-09-19:
Dear fahd7yazin,

Thanks for your message - I’m afraid I don’t yet understand on what basis you’d like to compare your two groups if you don’t include any of their responses in the analysis. Have I misunderstood something?

Best wishes,
Christoph

fahd7yazin, 2021-09-19:
Dear Dr Mathys,

Thank you for your response.

The main data I want to model with HGF is a continuous time series data, estimated from a group of subjects (group-averaged). This was akin to a continuous ‘rating’ data, rather than a classic behavioural response. My intent was to use a Bayes-optimal agent to model the belief trajectories for this time series. The estimated parameters would be compared to a neural data from a different set of subjects, where both groups observed the same stimulus with the latter group not giving any behavioural responses whatsoever (not even a rating). The logic being since the stimuli is exactly the same, the beliefs of one group should reflect the (Bayes-optimal) agents’ belief which is a reflection of the other group’s perceptual beliefs on this stimuli. It might be similar to these works https://doi.org/10.1523/JNEUROSCI.3069-19.2020, https://doi.org/10.1523/JNEUROSCI.3365-17.2018 where only a perceptual model is used (with decision/response module turned off).

chmathys, 2021-09-20:
Ok, thanks for this clarification - I think I understand. Since you’r not fitting agent responses, suggestion 1 from your original message looks appropriate. When comparing the different models, make sure to make your priors informative (i.e., tight) enough to give plausible (i.e., properly regularized, reasonably smooth) belief trajectories. If you blindly optimize LME, you'll be rewarded for loosening your priors too much and you'll get overfitted, jumpy trajectories.

Best wishes,
Christoph

P.S. If you feel reluctant to use informative, tight priors, that’s called the ‘Bayesian cringe’. To cure it, read this: https://statmodeling.stat.columbia.edu/2021/09/15/the-bayesian-cringe/

fahd7yazin, 2021-09-20:
Thanks for the clarification. 

The terminology of cringe seems apt for the hesitation that comes with using tight priors on any Bayesian modelling! (as far as frequentist critiques go)

Currently, what I did was create empirically informed models (informed by the neural results), but the mu, pi & sd still had wide priors while others had very tight priors as these coupling and scaling were key aspects. I optimized these according to the LME, but is it ok to use a more plausible model which has lower LME compared to a very wide priored model with large LME?

A few follow-up questions regarding the parameters: 

1)	If I want purely the (prediction) updates themselves with respect to a certain level (say level 3) in the hierarchy, the parameter of interest is _ud_ right? 
2)	What is the difference between precision weights (psi) and the full weights (wt) in the model output? 
3)	Is there a way to extract the outcome uncertainty parameter specifically from the learning rate?

Thanks

chmathys, 2021-09-22:
> Thanks for the clarification.
> The terminology of cringe seems apt for the hesitation that comes with using tight priors on any Bayesian modelling! (as far as frequentist critiques go)
> Currently, [...]
Yes - that’s exactly my point. LME can be gamed by choosing unrealistic (in terms of the prior predictive) priors. LME should only be used as a criterion *conditional* on a realistic prior predictive.
> A few follow-up questions regarding the parameters:
>  1. If I want purely the (prediction) updates themselves with respect to a certain level (say level 3) in the hierarchy, the parameter of interest is ud right?
Yes. Those are the updates with respect to the *prediction*: mu - muhat
>  2. What is the difference between precision weights (psi) and the full weights (wt) in the model output?
The weights in the update equations contain terms which are not precisions. psi only contains the precision ratio, ignoring the other terms (e.g., kappa^2, ...)
>  3. Is there a way to extract the outcome uncertainty parameter specifically from the learning rate?
In the standard continuous HGF, outcome uncertainty is a constant: 1/pi_u, i.e., the inverse of the outcome precision pi_u. In the original 2011 HGF paper, we used alpha for this, which is simply the inverse of pi_u (or pihat_u on my talk slides, which is the same). I always had a good reason to change the terminology, but in retrospect I probably should have stuck with my first choice to avoid confusion...

So the upshot is: you don’t need to extract outcome uncertainty anywhere, you already have it as a parameter.

fahd7yazin, 2021-09-22:
Thank you very much for the answers

closed 2021-09-22 by fahd7yazin