translationalneuromodeling/tapas #158: Quesion on the linear log-RT model developed for Marshall et al 2016

AlexAuteuil, 2021-09-20:
Dear Dr. Mathys, 

we are interested in the Marshall et al Plos Comp Biology (2016) paper and their use of the HGF "whatworld" functions to model behaviour and logRT in their task. We would like to do something similar but for the binary categorical model: we would like to explain logRT of participants' responses as a function of some of the HGF trajectories. 
What we don't understand is whether it would be possible to incorporate two types of responses: 
(I) 0/1 represeting choosing option A or B and modeled using the unit sigmoid response model (as in the examples in the hgf demo) and 
(II) logRT values of those 0/1 choices, which perhaps could be modeled using tapas HGF tapas_logrt_linear_binary.m function.

Is it possible to fit the perceptual model on the binary responses using the sigmoid response model and, in parallel, also fit the logRT? 
Otherwise, is there a publication that has used the tapas_logrt_linear_binary function (not the broaded whatworld as in Marshall et al 2016)?

Thank you

Best wishes,
Alex

chmathys, 2021-09-20:
Dear Alex,

Your idea makes sense and can easily be implemented. Just create a response model for both the binary choices and the logRTs, with a separate (and estimated) decision noise parameter for each of them. The magic of Bayesian inference will automatically lead to these noise parameters reflecting the information content of each of your two modalities of observations. The response model just needs to add the sums of the log-probabilities for both modalities and return this total sum.

Best wishes,
Christoph

StefanFraessle, 2021-09-21:
Dear Alex,

I just stumbled over your question and wanted to add some information to Christoph's excellent answer. More specifically, I just wanted to let you know that a PhD student of ours (Alexander Hess) is working on precisely what you describe - that is, a response model that fits both binary and continuous (RT) data simultaneously by combining two response model streams (just like Christoph has described). We are currently in the process of finalizing the project and hope to write this up soon, and then also make the code available for other people to use. So, we were wondering whether this might be of interest to you?

All the best,
Stefan

AlexAuteuil, 2021-09-21:
Dear Christoph and Stephan,

Thanks both. The answers by Christoph are very useful and I was thinking today about exploring the implementation. Stephan, if you and your PhD student are working on that, that is excellent and we would be interested in reading the work and using the associate code, when ready. I'm glad some people are working on this, as it could answer new exciting research questions.

We will get in touch via email in a few weeks, perhaps you have a preprint by then.

Best
Alex

AlexAuteuil, 2021-09-21:
Apologies, Stefan. Misswrote your name!

closed 2021-10-05 by chmathys

alexjhess, 2024-02-23:
Hi @AlexAuteuil,

It's quite a while since you asked your question but finally we have been able to upload a preprint (https://www.biorxiv.org/content/10.1101/2024.02.19.581001v1). Links to publicly available code and data are included. Hope this helps in case it is still of interest to you! :)

Cheers,
Alex