translationalneuromodeling/tapas #159: Appropriate model for specific multi-category physical stimulus values

lancegithub766, 2021-10-03:
Hi there,

I am trying to fit **tapas_hgf_whatworld** to tone sequences. I have seven distinct tones in this sequence

I have found that after setting _c.n_states = 7_ in **[tapas_hgf_whatworld_config.txt]**(https://github.com/translationalneuromodeling/tapas/files/7272329/tapas_hgf_whatworld_config.txt) I can only fit sequences with inputs c = [1,2,...7].
Inputting sequences with n distinct tone frequencies that are not [1,2, ...n] e.g. x= [856, 856, 1000, 1000, 1000, 1330, 1157, 1157, 1157,.....] with 7 unique values as per **[min 3, max 7, 26pct prob random.txt]**(https://github.com/translationalneuromodeling/tapas/files/7272332/min.3.max.7.26pct.prob.random.txt)
produces errors. Usually one of two: 

> - Arrays have incompatible sizes for this operation
> - Index in position 3 exceeds array bounds (must not exceed 7)

Using non-integer values e.g. [1.5,    2.2,    3.3,     4.4,    5.2,   6.7,   7.9] produces:

> - Index in position 3 is invalid. Array indices must be positive integers or logical values

(e.g. see [Lance_simulations-TonesCut_Copy.txt](https://github.com/translationalneuromodeling/tapas/files/7272457/Lance_simulations-TonesCut_Copy.txt), rename as .mlx to see first error )
So my question is, is **tapas_hgf_whatworld** only designed to work with inputs that are states in the form [1,2,..n]?
If so, I was thinking to convert my tones to [1,2,...7] but I think this will reduce the ability of the prediction errors spat out by the HGF to predict brain responses (e.g. MMN - mismatch negativiyy https://www.sciencedirect.com/topics/neuroscience/mismatch-negativity#:~:text=Mismatch%20negativity%20(MMN)%20is%20an,differs%20in%20frequency%20or%20duration ). Mapping the tone frequencies to [1,2,...7] seems maybe not so bad under the very strict assumption that the input tones are separated by perceptually equal intervals, but wouldn’t work at all for [1,102,1003, 1004...,1009] Hz where the difference between 1 and 102 isn’t the same as between 102 and 1003. In this case, I believe fitting Bayes optimal belief updating of means and variances using [1,2,...7] would no longer represent what’s happening in the brain for [1,102,1003, 1004...,1009].

If whatworld is not appropriate, I wonder if you can advise which model is? I am trying to answer questions like
i) How surprised should the participant be for each specific transition, and is there a correlation to MMN in the brain? How does this vary per the age of the participant? I am thinking to predict MMN for every stimulus transition, examining which HGF parameters have significant coefficients predicting age (or vice versa) like MMN = a*δ2 + b*ϵ3 + c*μ3   +ψ3,  and to try some non-linear functions using GAMS to predict MMN off PWPE + age 
ii) How different is the magnitude of the surprise as we move 1 perceptual unit away from the expectation vs 1.5 perceptual units etc. I want to examine things like tones whose pitch is 'accelerating' or has trends.
iii) Can we predict the latency of the mismatch response based on age, perceptual parameters etc.

As a second question, fitting the same sequence using ehgf doesn't produce an error, but produces no updating of u2, u3 (rename
[Lance_simulations - Tones - hgf ehgf.txt](https://github.com/translationalneuromodeling/tapas/files/7272453/Lance_simulations.-.Tones.-.hgf.ehgf.txt) to .mlx to see unless I set some of these not equal to zero
```

c.rhomu = [0, 0, 0];
c.rhosa = [0, 0, 0];
```
Config for that 
Thanks for your help!

lancegithub766, 2021-10-10:
Hi Christoph,

Thanks for your response. I remain a bit confused, though. I hope you can clarify the following

For your 2nd point, the authors of the paper you mention say

> `To estimate these parameters, we used the MATLAB function tapas_fitModel from the HGF toolbox (version 4.0), distributed as part of TAPAS (release version 1.6) with the _tapas_bayes_optimal_whatworld_ function`

i.e. also the whatworld model from the paper in your point (1) - albeit that that paper uses the tapas_logrt_linear_whatworld family of scripts (and they say they used the function _tapas_bayes_optimal_whatworld_config_ to determine priors)

But then don’t we run into the same problem you mention in your first point?

It’s exactly when I run this:
```
est_tones_hgf_whatworld = tapas_fitModel([], tones, 'tapas_hgf_whatworld_config', 'tapas_bayes_optimal_whatworld_config');
tapas_hgf_whatworld_plotTraj(est_tones_hgf_whatworld);
```
...that I get the problem where the inputs of tones must be [1,2,..n]

So if I understand correctly, whatworld only helps us understand transition probabilities. That is, it does not help us understand the next tone's expected tone frequency (or the distribution of this) - and thus the brain’s surprisal response? I am assuming the brain's surprisal response correlates not with the magnitude of the error in the transition probability to a tone (e.g. 26% minus the binary outcomes of what transition actually occurred: 0=did not transition, or 1=did transition to that), but based on the error in the frequency of the tone presented vs expected (e.g. the tone was 130Hz lower than expected, which is a t value of -1.2 with a normal distribution, say). It's this type of information that I want to be able to read off as the prediction error on each trial.

I suppose that given the tone transition probabilities, I could work out an _implied_ expected frequency for the next tone (sum of probabilities multiplied by the frequency for each tone it could transition to), but not the uncertainty in this, but this will be additional manual coding on top of whatworld??

If I am correct in the above then for my experiment, don’t I need to feed the HGF the actual tones [865, 865,... 1035,... ] etc?

If so, given that whatworld doesn’t allow this, which model can I use given that fitting the raw frequency values using ehgf produces no belief updating of u2, u3 as I described unless I change these 
```
c.rhomu = [0, 0, 0];
c.rhosa = [0, 0, 0];
```

Thank you in advance for your patient help in advance for this newbie!
Lance

BPaton, 2021-10-10:
Hi Lance,

Can't you as Christoph suggests, just encode your actual tone frequencies
as integers? That would give you transition probabilities between those
encoded frequencies (categories) and indirectly the expected tone (given
the current one)? And, being a little naive to the fine details of the what
world model, but wouldn't the precision weighted prediction errors for the
tone category (frequency) and the same pwPEs for the transition probability
happen at different levels in the hierarchy? At the lowest level you would
have violations that are only tone to tone (trial to trial) in their
extent, effectively something like, I expected 865 Hz but heard 1035 Hz, at
higher levels the longer term transition probabilities (iteratively summed
over many trials) may or may not be violated and hence lead to updates and
at higher levels how those transition probabilities change over time
(volatility)?

Thanks.

Bryan.

lancegithub766, 2021-10-12:
Hi Bryan, thanks for the response

I mentioned I could convert them to [1,2..n], but my issue is I'm not sure that doing so makes sense.

If the algorithm doesn't care about the stimulus values, then it compresses them down to simple categories and in so doing surely discards information? For example, provided the transition matrix was the same, it would then not care whether the tones are

- [100.1Hz, 100.2Hz, 100.3Hz] or
- [1Hz, 100Hz, 1000Hz] 

But surely a Bayesian optimal brain **does** care about this, i.e. not only about whether the tone predicted was in category A/B/C, but in how different the tones A, B, C are?
To take an extreme example, what if the options are [100Hz, 100Hz, 100Hz]. What is the significance of a prediction error if after the first tone we predict the 2nd tone but actually get the 3rd? They're all the same so there should be no surprise?

This is probably a lack of theoretical understanding on my part, but it's why I'm asking if I should use a different model (and following up on the errors I get when I do that).

As for things at the levels in the hierarchy, I think whatworld spits that out ok, (see attached my whatworld_analysis.txt which has PWPE at the 3rd level, for example), my issue is more understanding/conceptual interpretation as per above re: what say uncertainty at the third level means when we are looking at transition matrices rather than tone frequencies.
[whatworld_analysis.txt](https://github.com/translationalneuromodeling/tapas/files/7327164/whatworld_analysis.txt)

Thanks again,

Lance