translationalneuromodeling/tapas #160: Use PhysIO Toolbox to just denoise movement regressors

mdryzer1, 2021-10-08:
Hello,

I have been toying around with different movement denoising tools and would like to know if it is possible to just use the PhysIO Toolbox as a head motion denoiser. i.e., is there a way to run the batch editor without inputting cardiac or respiratory data?

Thank you.

mrikasper, 2021-10-08:
Dear Michael,

indeed, you can run PhysIO for the motion models only. For your reference, I have attached a version SPM batch file of the Philips ECG3T example where the physiological logfiles and models are removed and just the realignment parameters and motion outliers included. It's an `.m`-file, but I had to rename it to `.txt` for github.

All the best,
Lars
[philips_ecg3t_motion_only_spm_job.txt](https://github.com/translationalneuromodeling/tapas/files/7314050/philips_ecg3t_motion_only_spm_job.txt)

mdryzer1, 2021-10-14:
Hi Lars,

Thank you for the help! I got the code to work, but now I'm having issues using the produced motion regressors and outliers. Please see the attached image for more information.

[cid:81088780-20ba-4a20-a896-f31e29642b93]

This is an example of a GLM from my study. Each of the sessions should have an extra six columns for each of our 6 motion regressors; however, using the PhysIO regressors seems to remove these from my model. Any idea why this might be the case?

Thank you,

________________________

Michael Dryzer | PhD Candidate
Biomedical Engineering
Johns Hopkins University School of Medicine
716 N Broadway Baltimore, MD 21205
| E. ***@***.******@***.***>
https://www.linkedin.com/in/michaeldryzer/<https://nam02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fwww.linkedin.com%2Fin%2Fmichaeldryzer%2F&data=04%7C01%7Cmdryzer1%40jhmi.edu%7C8d05cab11043489034a608d97d0c017a%7C9fa4f438b1e6473b803f86f8aedf0dec%7C0%7C0%7C637678312823360329%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=YNYj6ISxu%2Fm8nquRxg0R04xFYpejnjhYDYUcgXhCvTA%3D&reserved=0>

      External Email - Use Caution

mrikasper, 2021-11-01:
Dear Michael,

you didn't misunderstand at all, my question about the GLM batch was a follow-up, because there was nothing conspicuous I could see in the PhysIO batch.

Looking at the `run_1st_level_job.m` file now, I also don't see anything that could explain the absence of the multiple regressors in the GLM specification.

But I don't have the image files, so I cannot run your script myself, and some of the logic is a bit nested to understand without having the data. Would you be able to provide the `matlabbatch` variable as it is returned at the end of the script, please?

Since you use `spm_select` to define the `multiple_regressors` files, this should also clarify whether they indeed exist where the script expects them 

```
    matlabbatch{1}.spm.stats.fmri_spec.sess(f + b).multi_reg = ... 
    cellstr(spm_select('FPList', fullfile(subject_path, 'func'), ['sub-' subjects{1} '_multiple_regressors_task-FatigueChoice_run-' num2str(f) '_bold.txt'])); 
```

All the best,
Lars

mdryzer1, 2021-11-02:
Hi Lars,

Please see the matlabbatch file you requested attached below. I looked at this file myself and found that all of the multiple regressors were blank, which would explain the absence of these regressors in the 1st-level GLM. I opened the script with the SPM batch editor and manually added the regressor files and got the GLM below (it has the right number of regressors for each session), so it would appear that there is an issue in my code concerning how the script selects the regressor files (as you suspected).

I'm looking now to see why the script isn't selecting the regressors and how to solve the issue. Thank you for all of your help thus far.

matlabbatch.zip: (https://github.com/translationalneuromodeling/tapas/files/7462067/matlabbatch.zip)

GLM with regressors: (https://user-images.githubusercontent.com/68439785/139881956-a7cacfb3-955b-4109-aa4f-029ac7b52407.png)

Sincerely,
Michael