translationalneuromodeling/tapas #163: using rDCM in task fMRI data

YounghyunO, 2021-11-10:
Dear all,

I have one question regarding the usage of the rDCM.
I was wondering whether it is appropriate to use rDCM to recover effective connectivity from the task fMRI data of the HCP dataset.
Specifically, I would like to use the preprocessed timeseries with MMP atlas (Glasser et al., 2016).
One question that I have regarding this is whether it would be appropriate to just use the timeseries only and to not use the input onset as regressors, which will result in assuming resting-state fMRI.

The reason I'm asking this is that, based on the assumptions of the rDCM I think it wouldn't be a big problem if I just use the timeseries data only. (However, there are other potential issues regarding the HRF)

Also, one question that might not be related to the rDCM toolbox (might as well be) is the usage of the resting state parcellations to do an analysis on task fMRI data. For example, as I mentioned I tried MMP atlas to do an analysis on HCP task dataset, however some type of the task data caused an error that had something to do with the collinearity, suggesting that the MMP atlas is parcellating brain regions too finely yielding redundant regions. Furthermore, there is numerous literature on the importance of right parcellation when conducting causality analysis. So I was thinking maybe this wasn't the right way to go, while at the same time there are many other studies doing the kind of analysis that I mentioned before (e.g., Deco et al., 2021; Global workspace revisited). 

Best wishes

StefanFraessle, 2021-11-15:
Hi Younghyun,

First of all, thank you very much for your interest in rDCM - I hope it proves useful for you.

Regarding your questions: In principle, you can do what you suggested and switch off all driving inputs. The model will allow you to do that. However, I would not advice to do so. The reason is that you are deliberately throwing away information that you know - namely the structure of the task (i.e., when certain stimuli were presented) and potentially even which regions should be driven by the input (e.g., visual regions if visual stimuli were presented). If you neglect this information, rDCM has to compensate for effects that might originate from driving inputs by means of connectivity parameters. This could give misleading results. As always: posterior parameter estimates are dependent on the model you have. Hence, the model (and the assumptions you put in there) do matter.

Also, switching off the driving inputs does not prevent the problem of rDCM with regard to region-specific HRFs. There is still the risk of inferring incorrect directionality in connections as a consequence of different HRFs in different regions. This is because the time series that you are using as predictors are the measured time series and thus are also subject to potential confounds of the HRF. 

Your question regarding the "right" parcellation scheme is indeed an important and difficult one. I completely agree that it is crucial to find the appropriate whole-brain parcellation scheme for the respective research question at hand. For instance, a resting-state based parcellation (e.g., Yeo, Schaefer) might be most suitable for rs-fMRI analyses. If you have task data, you might want to use a parcellation scheme that also takes into account task-fMRI (e.g., Glasser). Having said this, I feel that in your particular case, the error you obtain should not be related to using the Glasser (MMP) atlas. We have been using the Glasser parcellation for the HCP task data as well and did not run into any issues. The collinearity should really only be a problem if time series from different regions are identical - which is highly unlikely. Hence, my intuition would be that there is something else going wrong in your pipeline. 

In case you cannot find the problem, please feel free to send along an exemplary case where you get the error, so that I can have a look at it.

I hope this is helpful already.

With very best wishes,
Stefan

YounghyunO, 2021-11-17:
Thanks very much Stefan,

It indeed helped me resolve the questions.

Best wishes,
Younghyun

closed 2021-11-17 by StefanFraessle