translationalneuromodeling/tapas #171: Something wrong with regressor

zeyceln, 2022-02-07:
Hello, we seem to be getting TAPAS to work (to use PPG and Respiratory data to create regressors). However our first level matrix looks as attached. Do you know what the problem can be?

We are using Biopac data, and have created a matfile containing values for PPG and RESP per TR. Could that be the mistake? 

Thanks so much, I am really new to this and quite lost..

[image]
.

mrikasper, 2022-02-07:
Dear @zeyceln,

From your design matrix, it looks like the respiratory data could be compromised, because only the later regressors (to the right) experience the drop in amplitude for all but the first scan volumes. To investigate this further, could you please check and provide

1. The diagnostic plot for the input time series: "_Preproc: Diagnostics for raw physiological time series_"; This should show heart rate variability (and related issues with peak detection) as well as the breathing amplitude histogram (where a high peak at 0 would indicate issues such as breathing belt detachment)
2. The plot of the raw input time series "_Read-In: Raw Physiological Logfile Data_". Here you would see any problems with your recordings (e.g., very noisy or all-zero segments)
3. The matlab script or batch that you used to run PhysIO (`*.mat` or `.m`, depending on how you saved them in the batch editor), as well as the version (type `tapas_physio_version` in the matlab prompt) of PhysIO you are using (type `tapas_physio_version` in the matlab prompt).

Also, I am not quite sure what the BioPac data you provide to PhysIO looks like, i.e., what you mean by

> We are using Biopac data, and have created a matfile containing values for PPG and RESP per TR

Do you mean you do some extra processing after using th BioPac's software export to `.mat`?
TR would be on the order of seconds, so if you only provided one value per TR your signal would be aliased (we need sampling rates of 10s of Hz for the recordings). But if you just mean you exported BioPac files as `.mat` and work with columns such as

```
% 1 ECG100C
% 2 RSP100C
% 3 Digital input
% 4 Digital input   => could be trigger!
% 5 Digital input
% 6 C0 - Comb Band Stop Filter
% 7 C1 - Filter
```

you are probably fine.

I hope that helps.

All the best,
Lars

zeyceln, 2022-02-09:
Thank you for your answer. We tried with the new version and are getting similar output. 

Maybe we are doing something wrong while creating the Physiological input. We are using our best physio recording, so  it probably is not the data acquisition.

From Biopac, we are creating a mat file with 3 columns, 1 for Heart Rate, 1 for Respiration, and another for the TR trigger.

And the file format is as follows:
[image]

In our experiment we have 8 conditions in total. When creating the .mat file from .acq file, do we need to create a new file for each condition? Or groups of conditions (like FB pos, FB neg, FB ctrl)? 

Sorry if this is very trivial, but I really am not clear (we just created data points for each TR thinking it is data per scan that is needed).

Thanks in advance for your response,
Zeynep

mrikasper, 2022-02-09:
Dear Zeynep,

1. Thank you for clarifying the BioPac data export. The Matlab variables you show look very reasonable (about 1800 s of data, i.e., 30 minutes?), and you can probably feed the whole `.mat` file directly into PhysIO without splitting them into different files. Please do not resample the data to one value per TR, this will create aliasing (undersampling) issues. 
2. Regarding your 8 different conditions, do you mean they occur in different fMRI runs or do you have one consecutive 30 min experiment with interleaved condition blocks? If it is the former, you can still use one big input logfile (or decide to split them), but you have to run PhysIO for each individual run with the correct timing information (i.e., when the first or last scan volume of each run has happened relative to the physiological logfile).  You can do this 
    - manually by changing `log_files.relative_start_acquisition` (see `tapas_physio_new` or the Batch Editor for details) or 
    - use trigger information in the BioPac logfile (what are the 3 columns in the `data` variable, i.e., what is the content of `labels`
3. Could you provide the output plots from PhysIO I suggested in my previous post (item (1) and (2)), as well as the batch file (item (3)), please? This will facilitate any further debugging.

Thank you and all the best,
Lars

zeyceln, 2022-02-14:
Dear Lars,

Our experiment is in two runs. We split the data into runs and not for conditions- we have a 1st level design matrix  that looks good. We have been able to re-do the sampling with 10Hz. I am attaching a few pics below. I want to add the .mat file too, but I am not sure how to do it- cannot load it on github. Is there a specific part of the batch you would like a photo of to double check?
[image]
[image]
[image]

We are using Retroicor as we have ventricle activity during one of our conditions that is 90 seconds long. Unfortunately, there is not much change in that activity, if we have done everything right until now. 

Thank you in advance for all your support,
Zeynep

mrikasper, 2022-02-14:
Dear Zeynep,

Thank you for your response. Here are my suggestions

1. Could you please send the screenshots of the following figures: 
    - _"Read-In: Raw Physiological Logfile Data"_ (typically Figure 2, ideally zoom into the first 50s of the time course)
    - _"Preproc: Diagnostics for raw physiological time series"_ (typically Figure 5)
    - From your screenshots, while the design matrix doesn't look so suspicious any more, I think the cardiac RETROICOR regressor plot still has an atypical acceleration of its main frequency
2. I am still not sure what preprocessing pipeline you use to get the BioPac files, i.e., how you "re-do the sampling". Could you give some more detail here? Is that just an export option for BioPac to save as `.mat` file?
    - I am wondering why you chose "Custom" as your vendor format instead of "BioPac".
3. Regarding the batch file, you could either 
    - send me an e-mail (See [README](https://github.com/translationalneuromodeling/tapas/tree/master/PhysIO#readme) for the address), and then also include the problematic logfiles itself or 
    - save it from the Batch editor as `.m` file (change the "Save as Type") and rename this file to `.txt` before attaching it on GitHub. Please see the screenshots below.

[image]
[image]

I hope that helps!

All the best,
Lars

zeyceln, 2022-02-16:
Dear Lars,

Here are the two files you asked for, they indeed do not seem right:

[image]
[image]

We will send the batch tomorrow.

For the setting 'CUSTOM'-
 when we choose BIOPAC it expects (‘isi’, ‘isi_units’, ‘labels’, ‘start_sample’, ‘units’) to be filled in and we are not sure what to put in , with CUSTOM it is automatic. Is that a big problem?

Thanks so much for your response and patience,
Zeynep

mrikasper, 2022-02-17:
Dear Zeynep,

Thank you for the screenshots. Indeed, logging of both of your physiological timeseries seems to be stopping (all zeros) after about 10 min (~625s) in this example. 

1. From what I understand about your data, you have about 409 scans (or 1000s) in each fMRI run, and two runs. But from the length of your `.mat`-file variables, it seems that there is only one continuous BioPac file, correct? Are you splitting the Biopac file manually into two runs - if so, how do you decide where to split?
2. Regarding the `Custom` vs `BioPac_mat` setting - I think you wouldn't have to adjust anything. Just choose the `log_files.vendor` as  `BioPac_mat`  and put both `log_files.cardiac` and `log_files.respiration` as the same `.mat` file you get from the BioPac export. PhysIO identifies the relevant variables (that you listed and were in one of your workspace screenshots) automatically. If you get an error, it might be related to differing column labels (variable `labels`), but that should be easy to fix.
    - There is nothing wrong with using the `Custom` format, but for me it's harder to understand what happens until you arrive at the custom file vs the predefined Biopac-mat read-in. So it would be helpful if you run this one for comparison.

All the best,
Lars

zeyceln, 2022-02-25:
Dear Lars,

The Biopac files are split according to the number of scans per run through a matlab script.

We are trying to find the reason why the timeseries is stopping around 10min, as the Biopac file seems to have recorded the whole session in good condition. We will change the script we use to extract the Biopac data.

We have a question:

In the instructions it says : 'For Biopac and Siemens, sampling rate is read directly from logfile”
-	What is this logfile?
-	Is the Biopac file they are mentioning in *.acq or *.mat?

Thanks so much,
Best
Zeynep

mrikasper, 2022-03-10:
Dear Zeynep,

Thank you for sending the BioPac `.mat` files and my apologies for the delayed response. Grant season is coming up, and I have some deadlines to meet, I am afraid.

I did manage, however, to get a step further with your data: 

1. Indeed, choosing the “BioPac_mat” format in PhysIO works natively with your files, so you don’t have to split/convert it. I attached a relevant batch script, you might have to adapt the folder names.
    - I only checked it for the cardiac traces yet, are there respiratory traces recorded as well?
2. The next step you would have to do is get the timing right for both fMRI runs. I think all you have to do is 
    - setting the `log_files.relative_start_acquisition` differently for run 2. For the 2nd run it should be a positive number, sth. on the order of  `nVolumesInFirstRun*TR + delayBetweenRunsInSeconds + relativeStartAcquistionOfRun1InSeconds`
    - Check out the documentation of this parameter [here](https://github.com/translationalneuromodeling/tapas/blob/master/PhysIO/code/tapas_physio_new.m#L167)
3.  Additionally, you would have to set `logfiles.align_scan` to `first` instead of last, see [here](https://github.com/translationalneuromodeling/tapas/blob/master/PhysIO/code/tapas_physio_new.m#L188)
4. For the scan parameters (`sqpar`), I would set both `Nprep` and `Ndummies` to 0.

I hope that helps and 

All the best,
Lars
[batch_physio_biopac_direct_rename2m.txt](https://github.com/translationalneuromodeling/tapas/files/8225390/batch_physio_biopac_direct_rename2m.txt)