translationalneuromodeling/tapas #174: Example of rDCM question

StephDocTUM, 2022-03-02:
Dear all,
 
I have applied the rDCM method to dysconnective areas (dysconnective areas are defined in glioma patients, where the correlation values of the bold signals deviate over a certain standard deviation compared to a healthy cohort. Shown here as 1 to 3 (4 was not relevant)) applied in glioma patients.
Functionally in glioma patients not only the tumor area (number 2) is dysconnective or abnormal but also other areas in the ipsi- and contralateral hemisphere.
 
Now it is interesting how these dysconnective areas in the brain are causally related. 
 
  
[image]

 
I get the following matrix after applying the rDCM:
Now this tells me that region 1 to 3 is the most "effective" or causal?
 
 
[image]

 
Cheers,
Stephan

StefanFraessle, 2022-04-01:
Dear Stephan,

First of all, my apologies for the very long delay in responding - I just returned from parental leave and was quite busy catching up with things over the last few weeks.

Thank you very much for your interest in the rDCM toolbox and for giving the approach a try. I hope it proves useful for you.

Regarding your question: I do not really know much about glioma patients and, thus, can only provide information that is related to rDCM. First, in terms of the interpretation of the rDCM parameter estimates, I would refrain from using the term "causal". This is because one can debate (and many people have) whether the estimates of rDCM (and DCM in general) are causal in a strict mathematical sense. I would prefer to talk about "directed influences" between brain regions rather than "causal influences". 

Since the generative model of rDCM (and DCM in general) rests on a dynamical system, the connectivity parameters have to be interpreted as rate constants. Hence, mathematically, this means the values indicate how fast information flows from one region to another. This can be interpreted as the **strength** of a directed connection from one region to another. 

This means that the A parameter with the largest weight indicates the strongest connection in your network.

Hope this helps.

All the best
Stefan

StephDocTUM, 2022-04-19:
Dear Stefan,

Thanks a lot for your reply. What do you mean by A parameter?
Where would the output of the largest weight be in the TAPAS Toolbox?
From your paper, I was thinking this is visualized in the colors
[image]

Would it be possible to get access to the following course material:

https://www.tnu.ethz.ch/de/teaching/hs-2021/methods-models-for-fmri
(especially about DCM)