translationalneuromodeling/tapas #176: Warning: First volume with missing slice events (219: 1 instead of 31 found)

NinetLab-Redux, 2022-03-20:
Dear Lars,

Thank you for creating and maintaining such an amazing toolbox. 
I'm new to its use and I got stuck at the above warning. I read the threads on multiband and I believe I input the right nslices parameter. 
I'm confused as to why in the Threshold Gradient it detects 219 volume events when on the command line it correctly identifies the 210 + 4 dummies (see error log).
Any pointers as to where I'm not inputting the right parameters would be much appreciated. 
best regards,
fvr

[ErrorLog.txt](https://github.com/translationalneuromodeling/tapas/files/8310676/ErrorLog.txt)
[Sub-003_job.m.zip](https://github.com/translationalneuromodeling/tapas/files/8310677/Sub-003_job.m.zip)
[image]
[image]

NinetLab-Redux, 2022-03-26:
Update: I got it to work :)  Our lab is aiming to put together a video tutorial for training purposes of new lab members and, with your approval first, we could share it with this community? :) 
[Figures_Output 10.pdf](https://github.com/translationalneuromodeling/tapas/files/8355688/Figures_Output.10.pdf)

mrikasper, 2022-04-04:
Dear Fidel,

Great you got it to work, thank you for the persistence and sorry for my late reply! The setting of the `scan_timing` parameters for the Philips logfiles is always a bit cumbersome, because it is based on the undersampled loggging of the gradient waveform (e.g., EPI) in the physiological logfile. While there is almost always sufficient periodicity to identify the timing of each slice acquisition, this usually needs some iterations (Looking at the `Sync: Thresholding Gradient...` figure after each error and following the suggestions in the error message) - see also this [FAQ](https://gitlab.ethz.ch/physio/physio-doc/-/wikis/FAQ#12-philips-i-would-like-to-use-the-gradient-log-for-timing-synchronization-but-how-do-i-set-the-thresholds) for a short description.

So, indeed, having a video tutorial on that would be awesome, and I am happy to provide some feedback. We also have a TAPAS youtube channel where such videos could be posted in principle. 

The good news is that these `scan_timing` parameters should be consistent for all subjects acquired with the same acquisition sequence, so hopefully you have to go through these steps only once per neuroimaging study.

After our bilateral meeting, I revised the matlab job that now runs through smoothly for me, I sent it to you via e-mail last week, and now insert it here as future reference for the user community.

Let me know how I can be of further help!

All the best,
Lars

```
matlabbatch{1}.spm.tools.physio.save_dir = {'C:\Users\kasperla\polybox\Shared\PhysioUserData\gh176-philips-slice-events-fidel\physIO_out_LK'};
matlabbatch{1}.spm.tools.physio.log_files.vendor = 'Philips';
matlabbatch{1}.spm.tools.physio.log_files.cardiac = {'C:\Users\kasperla\polybox\Shared\PhysioUserData\gh176-philips-slice-events-fidel\SCANPHYSLOG20220216124638_SE.log'};
matlabbatch{1}.spm.tools.physio.log_files.respiration = {'C:\Users\kasperla\polybox\Shared\PhysioUserData\gh176-philips-slice-events-fidel\SCANPHYSLOG20220216124638_SE.log'};
matlabbatch{1}.spm.tools.physio.log_files.scan_timing = {''};
matlabbatch{1}.spm.tools.physio.log_files.sampling_interval = 0.00201612903225806;
matlabbatch{1}.spm.tools.physio.log_files.relative_start_acquisition = 0;
matlabbatch{1}.spm.tools.physio.log_files.align_scan = 'last';
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Nslices = 31;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.NslicesPerBeat = [];
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.TR = 2;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Ndummies = 4;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Nscans = 210;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.onset_slice = 1;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.time_slice_to_slice = 0.032258064516129;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Nprep = [];
matlabbatch{1}.spm.tools.physio.scan_timing.sync.gradient_log.grad_direction = 'z';
matlabbatch{1}.spm.tools.physio.scan_timing.sync.gradient_log.zero = 1900;
matlabbatch{1}.spm.tools.physio.scan_timing.sync.gradient_log.slice = 2000;
matlabbatch{1}.spm.tools.physio.scan_timing.sync.gradient_log.vol = [];
matlabbatch{1}.spm.tools.physio.scan_timing.sync.gradient_log.vol_spacing = 0.1;
matlabbatch{1}.spm.tools.physio.preproc.cardiac.modality = 'ECG_WiFi';
matlabbatch{1}.spm.tools.physio.preproc.cardiac.filter.no = struct([]);
matlabbatch{1}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.min = 1;
matlabbatch{1}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.file = 'initial_cpulse_kRpeakfile.mat';
matlabbatch{1}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.max_heart_rate_bpm = 90;
matlabbatch{1}.spm.tools.physio.preproc.cardiac.posthoc_cpulse_select.off = struct([]);
matlabbatch{1}.spm.tools.physio.preproc.respiratory.filter.passband = [0.01 2];
matlabbatch{1}.spm.tools.physio.preproc.respiratory.despike = false;
matlabbatch{1}.spm.tools.physio.model.output_multiple_regressors = 'multiple_regressors.txt';
matlabbatch{1}.spm.tools.physio.model.output_physio = 'physio.mat';
matlabbatch{1}.spm.tools.physio.model.orthogonalise = 'none';
matlabbatch{1}.spm.tools.physio.model.censor_unreliable_recording_intervals = false;
matlabbatch{1}.spm.tools.physio.model.retroicor.yes.order.c = 3;
matlabbatch{1}.spm.tools.physio.model.retroicor.yes.order.r = 4;
matlabbatch{1}.spm.tools.physio.model.retroicor.yes.order.cr = 1;
matlabbatch{1}.spm.tools.physio.model.rvt.yes.method = 'hilbert';
matlabbatch{1}.spm.tools.physio.model.rvt.yes.delays = 0;
matlabbatch{1}.spm.tools.physio.model.hrv.yes.delays = 0;
matlabbatch{1}.spm.tools.physio.model.noise_rois.no = struct([]);
matlabbatch{1}.spm.tools.physio.model.movement.yes.file_realignment_parameters = {'C:\Users\kasperla\polybox\Shared\PhysioUserData\gh176-philips-slice-events-fidel\rp_Sub-003_WIP_VBRAIN_rFMRI_2.5_S2MB2_20220216121300_601.txt'};
matlabbatch{1}.spm.tools.physio.model.movement.yes.order = 12;
matlabbatch{1}.spm.tools.physio.model.movement.yes.censoring_method = 'FD';
matlabbatch{1}.spm.tools.physio.model.movement.yes.censoring_threshold = 1;
matlabbatch{1}.spm.tools.physio.model.other.no = struct([]);
matlabbatch{1}.spm.tools.physio.verbose.level = 2;
matlabbatch{1}.spm.tools.physio.verbose.fig_output_file = 'figures';
matlabbatch{1}.spm.tools.physio.verbose.use_tabs = false;

```

negorova, 2024-03-20:
Hello, I seem to be getting a similar problem as this thread. 

First I get a warning: Warning: First volume with missing slice events (800: 83 instead of 84 found)

and then an error: Error using tapas_physio_get_sample_points
scan 85: only 83 slice scan events. Cannot resample to slice 84

I have 84 slices x 800 volumes in each file. For some participants the same analysis runs without error (identical scripts), and for some (about a third) this error occurs and it all fails. I am not sure why.

Please let me know what logs or files I should send. Thanks in advance for your help!

negorova, 2024-03-20:
Dear Lars, thanks so much for your quick response. 

I specify 84 slices. I have attached the Siemens logs, batch and error log in the attachment. Note again that some of the subjects run with this set up successfully.

Thanks for your help!

Natalia

[logs.zip](https://github.com/translationalneuromodeling/tapas/files/14672826/logs.zip)

negorova, 2024-04-03:
> Dear @negorova<https://github.com/negorova>,
>
> While it sounds similar, this might actually be a different issue. Which scanner vendor are the logfiles from (Siemens/Philips)? If it is Siemens, one potential reason could be a weird interaction with the multiband acquisition as in this issue<https://github.com/translationalneuromodeling/tapas/issues/258#issuecomment-1941983542>.
>
> What do you specify as the number of slices for your PhysIO script?

Dear Lars,

I have added a bit more information about the errors, and was wondering if you could please take a look at the logs I had attached. I am still unable to resolve the issue. Thanks for your help!

Kind regards,
Natalia

Natalia Egorova-Brumley, PhD | Associate Professor
Australian Research Council Future Fellow

Melbourne School of Psychological Sciences
Office 804, Redmond Barry Building
The University of Melbourne, Victoria 3010 Australia
T: +61 3 9035 7053  E: ***@***.***
W: www.egorovanatalia.com

Pain and Cognition Neuroimaging Lab
W: https://psychologicalsciences.unimelb.edu.au/research/msps-research-groups/pain-and-cognition-neuroimaging-lab

I acknowledge the Traditional Owners of the land on which I work, and pay my respects to the Elders, past and present.

[PastedGraphic-1.png]

RaphaSchl, 2024-05-02:
Hi,

I have exactly the same issue as Natalia, and was wondering what could be done about it. 

I get the following error : 
`
Warning: First volume with missing slice events (212: 316 instead of 317 found)
`
and further down 
`
Error using tapas_physio_get_sample_points
scan 1: only 316 slice scan events. Cannot resample to slice 317
`

My images are made up of 212 volumes, and 317 slices. For other subjects in my dataset, TAPAS runs smoothly.
If you have any tips on what to look out for, that would be great.

Best,
Raphaëlle

mrikasper, 2024-05-02:
Dear Raphaëlle,

The issue looks indeed similar, but it's actually a slightly different one for your single-band acquisition compared to the multi-band one above. I am working on a fix, but haven't completed it yet. Would you also be able to share an example dataset and matlabbatch/script to reproduce the error, please?

Thank you for your help!

Lars

RaphaSchl, 2024-05-02:
Dear Lars,

Thank you for your quick answer ! In [this folder ](https://filesender.renater.fr/?s=download&token=2f798427-032c-41f5-b963-9e4dd182cbd1) you should be able to find all you need to reproduce the error.

In the script, you will see I excluded the PULS signal as I noticed it was flat, and planned on running RETROICOR only with the RESP signal. It might be surprising, but it is for a spinal fMRI analysis - for a previous study I was able to apply the toolbox to all my data and had it working like a charm :)

Please do keep me posted, and thank you for your time !
Cheers,
Raphaëlle

RaphaSchl, 2024-07-08:
Hi @mrikasper,
Were you able to resolve the issue we discussed previously?
Best,
Raphaëlle