translationalneuromodeling/tapas #177: rDCM Experimental Inputs

JBarsotti, 2022-03-21:
Sorry if this is a stupid question, but could you elaborate a bit on what the U input is to the **rDCM function tapas_rdcm_model_specication()**? More specifically, what does the sparse matrix U.u represent?

StefanFraessle, 2022-04-01:
Dear John,

First of all, my apologies for the long delay in responding - I just returned from parental leave and was quite busy catching up with things over the last few weeks.

Thank you very much for your interest in the rDCM toolbox and for giving the approach a try. I hope it proves useful for you.

Regarding your question, `U` represents the experimental inputs that, if you are working with task-based fMRI, can enter the network and perturb the neuronal states directly. In `tapas_rdcm_model_specification.m`, U is a structure that has two fields. 

1. `U.u` comprises the experimental inputs. This is a (16*Nr_Y) x Nr_U matrix, where Nr_Y is the number of data points (i.e., fMRI scans) and Nr_U is the number of experimental inputs. You can think of U.u as similar to the design matrix in a first-level GLM analysis. 
2. `U.name` comprises the name of the experimental inputs (e.g., U.name = {"faces", "houses"}) 

If you are working with resting-state fMRI data, there are no experimental inputs and you do not need to specify anything. Hence, when constructing your model using `tapas_rdcm_model_specification.m`, you simply need to provide an empty argument for U:

`[ DCM ] = tapas_rdcm_model_specification(Y, [], args);`

The toolbox will then correctly identify this as a resting-state model and make the respective adjustments.

Hope this helps.

All the best
Stefan

JBarsotti, 2022-04-01:
Thank you so much for the reply! That helps a lot!

closed 2022-04-04 by StefanFraessle