translationalneuromodeling/tapas #178: max operator in eHGF update for highest level precision for numeric reasons?

Murdugan, 2022-04-01:
I compared the update equations for the eHGF to the regular HGF and stumbled upon the max operator in the precision update for the highest level precision in line 191 of the tapas_eHGF_binary function and was wondering where the max comes from:

pi(k,l) = pihat(k,l) +max(0, 1/2 *ka(l-1)^2 *ww*(ww +rr*dd));

Is this just a numerical trick or does it fall out of the derivation of the new update equations?

Thank you so much in advance.

chmathys, 2022-04-01:
Hi Murdugan,

This reflects the fact that, logically, an observation on a variable cannot increase uncertainty about it. Therefore, the posterior precision pi cannot be less than the precision of the prediction pihat. This principle trumps what the update equation says since that is only approximate.

This will also be explained in our forthcoming technical report.

Best wishes,
Christoph

Murdugan, 2022-04-04:
Thank you so much for the quick response. I am looking forward to the report.

closed 2022-04-04 by Murdugan