translationalneuromodeling/tapas #179: interpretation of terms in high-level precision update in the eHGF

Murdugan, 2022-04-01:
I was wondering, if the ww, rr, dd, vv, and pimhat for the highest-level precision updates in the tapas_eHGF_binary function:

        % Ingredients of the precision update which depend on the mean
        % update
        vv = t(k) *exp(ka(l-1) *mu(k,l) +om(l-1));
        pimhat = 1/(1/pi(k-1,l-1) +vv); 
        ww = vv *pimhat;
        rr = (vv -1/pi(k-1,l-1)) *pimhat;
        dd = (1/pi(k,l-1) +(mu(k,l-1) -muhat(k,l-1))^2) *pimhat -1;
                
        pi(k,l) = pihat(k,l) +max(0, 1/2 *ka(l-1)^2 *ww*(ww +rr*dd));

carry specific meanings that one could include when describing these updates similar to the uncertainty partitions in the original HGF such as informational and environmental uncertainty as discussed in Mathys et al. (2014) or are some of these terms just summarized for convenient coding?

Thank you so much in advance.

chmathys, 2022-04-01:
They were introduced for coding convenience - it would of course be fascinating to see whether they can be given a substantive interpretation!

Murdugan, 2022-04-04:
Thanks a lot!

closed 2022-04-04 by Murdugan