translationalneuromodeling/tapas #186: PhysIO user and developer meeting at ISMRM

mrikasper, 2022-04-13:
Dear PhysIO users and developers,

I will be attending the ISMRM conference in London this year (not OHBM though). If you are also coming and would like to meet in person to discuss the direction of PhysIO and your use cases for future development, this would be a great opportunity.

I have tagged our external contributors here: @DanielHoffmannAyala @benoitberanger @jsheunis @nbuergi @alexsayal @nbeliy, but of course anyone interested in PhysIO would be very welcome to attend.

Please let me know if this is of interest (either here or via e-mail, see README), and I can organize a little get-together.

All the best,
Lars

jsheunis, 2022-04-13:
@mrikasper this sounds like a great idea. Unfortunately I won't be at ISMRM this year. But I will be in Glasgow for OHBM if anyone wants to meet up.

benoitberanger, 2022-04-14:
Hello,

Same for me, I will probably be attending OHBM, but not ISMRM.
But is there are discussions about PhysIO developments, I'm in ! 

Best,
Benoît

jaetzel, 2022-04-20:
I am planning to be at ISMRM (my first ISMRM and first in-person conference since covid!) and would like to meet. We haven't been using PhysIO recently, but are still collecting physio during our DMCC fMRI sessions and I hope to analyze it properly at some point.

mrikasper, 2022-05-05:
Dear all,

Thank you so much for your feedback, that's great!

I would suggest 3 things:

1. Since many of you are at HBM, maybe you can use this thread to schedule a meeting between all of you, and I try to join virtually (no promises, I may be on parental leave). There is an educational session at OHBM on June 19th organized by Stefano Moia @smoia on physiological fluctuations with a great line-up (_Physiologic fMRI Signals: Friend or Foe?_) that could be a natural meeting point.
    - BTW: I have created a [3 minute intro video to PhysIO](https://youtu.be/W6Z4woK1gMA) that still has to be trimmed for the educational session, let me know what you think!
2. Let's have a virtual meeting in the aftermath of the conference get-togethers (maybe July or after the summer break) to discuss about the roadmap for PhysIO. I am particularly keen to learn what features you are currently missing and/or are willing to develop, but also want to brainstorm any ideas on how to properly attribute external contributions. I know that putting your names in README and Contributors-files doesn't really reflect how much work you put into development, especially for traditional evaluations of your research output.
4. @jaetzel Joset, do you have a presentation at ISMRM? That would be a natural time and place to meet. Otherwise, I can probably reach you on the ISMRM app? (or e-mail?). If you are there for the educational weekend already, I am pretty free to meet then.

All the best, and looking forward to meeting you soon,
Lars

smoia, 2022-05-05:
Hi all!
Got pinged so I guess I'll join the conversation :wink: 
I'm going to be at ISMRM and OHBM _and_ OHBM Brainhack (which is a very fun experience you should consider registering to, it runs June 16th-18th in Glasgow), and be happy to meet any PhysIO user and chat a bit!
I have a presentation and a poster at ISMRM that are somewhat physiolo

See you in the upcoming

mrikasper, 2022-06-09:
Dear PhysIO users/developers/friends @DanielHoffmannAyala @benoitberanger @jsheunis @nbuergi @alexsayal @nbeliy @smoia @jaetzel,

I realize the virtual part of OHBM has already started and soon some of you will meet in person at OHBM. I will be on parental leave for a few weeks starting tomorrow. 

It was great to see some of you at ISMRM and exchange ideas and experiences, and I realize this is best done in person. So I asked Johanna Bayer @likeajumprope - whom you might know from the @brainhackorg or @ohbm Hackathon - whether she could meet with you during the conference to continue the conversation. Johanna and I worked on a project in Zurich together previously, and she might come to Toronto for a few months to work on open science software development with me, so I thought that would be a great way to introduce her to the PhysIO community.

Maybe she can reach out to you to find time for a coffee and chat about missing features/use cases/communication around PhysIO, and brainstorm about properly attributing external contributions. I recently had a first go at a "Ten Simple Rules for PhysIO-logical Noise Correction in fMRI" talk, and this could develop into a tutorial/white paper with all of the external contributors as co-authors. Just a thought to inspire discussion.

All the best,
Lars

smoia, 2022-06-09:
Hi all!
I was also planning on submitting a proposal for an emergent session with physiopy to continue our best practices meeting of June. If you want we can pull the two occasions together - I was about to send out the call for topic preferences to the community and I can open it to you all as well!

jsheunis, 2022-06-09:
I'm in 👍 😁

mrikasper, 2022-06-10:
That sounds great, @smoia! A pity I cannot be there.

Cheers,
Lars

smoia, 2022-06-10:
It will have online interaction, but to be fair I think you'll have other things on your mind that week :sweat_smile:

smoia, 2022-06-12:
Hi all!
Please find [here](https://forms.gle/QV5ar56esfimL1vj8) the form to vote for the topic of our meeting - and eventually give your availability for a drink meetup. Please fill in by the June 14th at 23.59PM, and leave your email on the last page.

I can't guarantee 100% that the emergent session I'll propose and will constitute the meeting will be picked up, but I'll try as hard as I can to make it happen!

smoia, 2022-06-19:
Hi all, our emergent session has been confirmed for Tuesday evening at 17.45. I'd be happy to see you there, and if you have suggestion on topics to cover - or you want to help with the hosting, please let me know!