translationalneuromodeling/tapas #188: tapas_simModel in HGF 6.0 not compatible with tapas_rw_binary

dsstolz, 2022-05-06:
I have noticed that new lines have been inserted (166-176) in tapas_simModel, serving to:
% Check inferred states for NaN values (due to numerical problems when taking log)

This code requires the existence of r.traj to contain muhat and sahat, which is not the case for, e.g., tapas_rw_binary, which produces v instead of muhat. Therefore, using tapas_simModel from version 6.0 is incompatible with rw models (and possibly others that to not produce muhat and sahat).

alexjhess, 2022-05-09:
Dear @dsstolz,

Thank you for pointing this out! We are working on a solution and will notify you once we have fixed it.

Best wishes,
Alex

chmathys, 2022-05-16:
A bugfix release will happen in the coming days

chmathys, 2022-05-17:
[Bugfix release](https://github.com/translationalneuromodeling/tapas/releases/tag/v6.0.1)

closed 2022-05-17 by chmathys