translationalneuromodeling/tapas #197: Multiple Realignment Parameter Files

alireza-kr, 2022-06-12:
Hello

I tried using PhysIO Toolbox to employ additional regressors (**CompCor**) in my preprocessing pipeline. My experiment has _three runs_, so it has three different _realignment parameters files_ resulting from SPM preprocessing. I used the following script from PhysIO Toolbox in my pipeline, and I want to have realignment parameters files.

```
matlabbatch{stage}.spm.tools.physio.save_dir = {mytempdir};
matlabbatch{stage}.spm.tools.physio.log_files.vendor = mri.spm.vendor;
matlabbatch{stage}.spm.tools.physio.log_files.cardiac = {''};
matlabbatch{stage}.spm.tools.physio.log_files.respiration = {''};
matlabbatch{stage}.spm.tools.physio.log_files.scan_timing = {''};
matlabbatch{stage}.spm.tools.physio.log_files.sampling_interval = [];
matlabbatch{stage}.spm.tools.physio.log_files.relative_start_acquisition = 0;
matlabbatch{stage}.spm.tools.physio.log_files.align_scan = 'last';
matlabbatch{stage}.spm.tools.physio.scan_timing.sqpar.Nslices = mri.spm.nslices;
matlabbatch{stage}.spm.tools.physio.scan_timing.sqpar.NslicesPerBeat = [];
matlabbatch{stage}.spm.tools.physio.scan_timing.sqpar.TR = mri.spm.TR;
matlabbatch{stage}.spm.tools.physio.scan_timing.sqpar.Ndummies = 0;
matlabbatch{stage}.spm.tools.physio.scan_timing.sqpar.Nscans = mri.spm.nvols;
matlabbatch{stage}.spm.tools.physio.scan_timing.sqpar.onset_slice = mri.spm.refslice;
matlabbatch{stage}.spm.tools.physio.scan_timing.sqpar.time_slice_to_slice = [];
matlabbatch{stage}.spm.tools.physio.scan_timing.sqpar.Nprep = [];
matlabbatch{stage}.spm.tools.physio.scan_timing.sync.nominal = struct([]);
matlabbatch{stage}.spm.tools.physio.preproc.cardiac.modality = 'ECG';
matlabbatch{stage}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.min = 0.4;
matlabbatch{stage}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.file = 'initial_cpulse_kRpeakfile.mat';
matlabbatch{stage}.spm.tools.physio.preproc.cardiac.posthoc_cpulse_select.off = struct([]);
matlabbatch{stage}.spm.tools.physio.model.output_multiple_regressors = 'mr_.txt';
matlabbatch{stage}.spm.tools.physio.model.output_physio = 'physio.mat';
matlabbatch{stage}.spm.tools.physio.model.orthogonalise = 'none';
matlabbatch{stage}.spm.tools.physio.model.censor_unreliable_recording_intervals = false;
matlabbatch{stage}.spm.tools.physio.model.retroicor.no = struct([]);
matlabbatch{stage}.spm.tools.physio.model.rvt.no = struct([]);
matlabbatch{stage}.spm.tools.physio.model.hrv.no = struct([]);
matlabbatch{stage}.spm.tools.physio.model.noise_rois.yes.fmri_files = {'fmri1.nii';'fmri2.nii','fmri3.nii'};
matlabbatch{stage}.spm.tools.physio.model.noise_rois.yes.roi_files = {'rc2anat.nii';'rc3anat.nii'};
matlabbatch{stage}.spm.tools.physio.model.noise_rois.yes.force_coregister = 'No';
matlabbatch{stage}.spm.tools.physio.model.noise_rois.yes.thresholds = 0.9;
matlabbatch{stage}.spm.tools.physio.model.noise_rois.yes.n_voxel_crop = 1;
matlabbatch{stage}.spm.tools.physio.model.noise_rois.yes.n_components = 5;
matlabbatch{stage}.spm.tools.physio.model.movement.yes.file_realignment_parameters = {'rp_fmri1.txt';'rp_fmri2.txt';'rp_fmri3.txt'};
matlabbatch{stage}.spm.tools.physio.model.movement.yes.order = 6;
matlabbatch{stage}.spm.tools.physio.model.movement.yes.censoring_method = 'none';
matlabbatch{stage}.spm.tools.physio.model.movement.yes.censoring_threshold = 0.5;
matlabbatch{stage}.spm.tools.physio.model.other.no = struct([]);
matlabbatch{stage}.spm.tools.physio.verbose.level = 2;
matlabbatch{stage}.spm.tools.physio.verbose.fig_output_file = 'physio.fig';
matlabbatch{stage}.spm.tools.physio.verbose.use_tabs = false;
```
However, when I ran the above script, I got the following error:
> Item 'Realignment Parameter File', field 'val': Number of matching files larger than max allowed, keeping 1/3 files.

It seems that I can't add all realignment parameter files, and I am allowed to have only _one file_. What can I do to have all realignment parameter files?

benoitberanger, 2022-06-13:
Hi @alireza-kr,

You **cannot** process several fmri files in a single PhysIO job.
1 job (`matlabbatch{idx}`) = 1 fmri run to process.

Best,
Benoît

alireza-kr, 2022-06-24:
Thank you.

I created one batch job per run and there was no problem.

closed 2022-06-24 by alireza-kr