translationalneuromodeling/tapas #198: Changing StartTime in .json file to 0 produces no respiratory regressors & other respiratory regressor estimation problems

lbanellis, 2022-06-15:
Hi & thank you for your amazing package.

I am fairly new to fMRI and the tapas/PhysIO toolbox so apologies if this a stupid question. I am using an edited version of the bids_ppu3t_matlab_script.m and I noticed when I change the StartTime parameter in the .json file to 0 (to avoid simulated samples pre respiration time series & so the plots start at 0 instead of the actual StartTime of the scan) the respiratory regressors become zero's (see figures: first set of figures at actual StartTime, second at 0). Should the StartTime influence the estimated respiratory regressors? 

Actual StartTime:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

StartTime at 0:
[image]
[image]
[image]
[image]
[image]
[image]

Furthermore, only for some subjects are respiratory regressors estimated (even if the StartTime is set to actual StartTime of the scan). For example, the above first set of figures with estimated respiratory regressors are from this raw respiratory data:
[image]

But for the raw respiratory data from these subjects, no respiratory regressors are estimated:
[image]
[image]

Do you have any advice for how I could preprocess the respiratory data so respiratory regressors are estimated?

Attached is my code and data in case it helps.

Thanks in advance! 

[PhysIO.zip](https://github.com/translationalneuromodeling/tapas/files/8908840/PhysIO.zip)

mrikasper, 2022-06-18:
Dear Leah,

Thank you so much for using PhysIO and providing such a detailed bug report. I am currently on a break following the arrival of our latest family member, but will get back to you the week after next - let me know if resolving your questions is more time-critical.

In the meantime, I was wondering how your BIDS files were created. The `StartTime` should be determined by your converter - it tells you when the logging started relative to the scanning, so indeed it would alter the computed regressors. Maybe you can give a bit more detail on why you simulate extra data and adjust `StartTime`?

Internally, PhysIO uses, for example, DICOM and phys log file timestamps to sync scan and peripheral recording data, so there should be no need to adjust this. If your original data is in one of the formats PhysIO supports, you might wanna try to read in that data directly instead of the BIDS-converted one. Sadly, at the moment, the BIDS standard is not free of information loss compared to the scanner or peripheral device vendor data.

All the best,
Lars

mrikasper, 2022-06-18:
PS: Regarding the odd respiratory time series from which no respiratory regressors could be created, again it seems as if the conversion to BIDS might be the issue. If you look at your `.tsv` files (the raw tab-separated value column data), you see that the values are saved as floats, e.g., `2.4743e+06`. but only with a precision of 4 digits after the dot. It can very well be that the actual amplitude changes are at the 5th or 6th digit and you end up with information loss due to using the wrong dynamic range of data representation. Typically, the values I see for physiological traces are not in the millions...

All the best,
Lars

lbanellis, 2022-06-22:
Dear Lars

Thank you for your quick response & congratulations on your new family member! (& apologies for disturbing you during this time). 

The raw data is from a BrainVision system (.vhdr files) with triggers sent from the scanner. We converted the data to a BIDS formatted .tsv file with StartTime reflecting the time of the first TR trigger (i.e., start of resting state or task for example). Should StartTime reflect something else? (I was only adjusting StartTime to experiment with the plotting as I noticed the example BIDS data started much earlier - & therefore the plots didn't add simulated data before the StartTime). However, I won't change StartTime if what I described is correct.

I imported the raw BrainVision .vhdr file and it seems to be in the same scale:
[image]

The respiration and cardiac data are acquired with electrogastrography, not sure if that influences the scale. Would you suggest rescaling the data?

Thank you for your help in advance.

Best wishes,
Leah

mrikasper, 2022-06-27:
Dear Leah,

Thank you for the follow-up, I hope I will be more responsive now.

1. Regarding the `StartTime` parameter, I think using the scanner trigger signal in your vendor recording file is perfectly sound - I am just unsure whether you specified the right sign for the value:
    - As per BIDS specification, the start of the physiological logging is stated relative to the first volume scan start, i.e., if physiological logging starts before the first scan volume, this number should be negative.  
    - This is the opposite convention that PhysIO uses internally for timing parameters, e.g., `log_files.relative_start_acquisition` defines the scan acquisiton relative to the phys log start (i.e., if physiological logging starts before the first scan volume, this number would be positive). 
    - If you just use BIDS, make sure you are consistent with the first convention and set the `log_files.relative_start_acquisition` parameter to 0.
   - I can see you specified a positive value in all your BIDS files, which would mean that the physiological recording started *after* the fMRI scan was started, and considerably so (720s would be 12 minutes). This is why PhysIO tries to simulate the missing data it needs for the first twelve minutes of fMRI scanning. This makes the estimation so slow, since this algorithm is usually just employed for gaps of a few (tens of) seconds. Could you double-check, please, if these signs are correct, and let me know/send the updated `.json` files? 
   - NB: Per [BIDS specification](https://bids-specification.readthedocs.io/en/stable/04-modality-specific-files/06-physiological-and-other-continuous-recordings.html), you don't have to specify and `EndTime`, we won't use it in PhysIO.
2. Regarding the respiratory traces and their small dynamic range/the zero-value respiratory RETROICOR regressors
    - I am not familiar with BrainVision's electrogastrograhy system and that might very well be the reason that the data appeared uncommon to me. 
    - The scaling in the millions should per se not be problematic, but the loss of precision in the digitization is (it basically looks as if you have only 10 different values in your time series, which severely limits SNR). Is there any way to adapt the dynamic range for these channels in BrainVision? 
    - I also noticed that you attached a file name containing a "filtered" label for one subject (`sub-0022`) and raw (?) data for the other two. In both cases, I see some strong oscillations just before the start of the first scan or at the end of the time series (see screenshot). Is there some filtering performed for all subjects that could explain this oscillation?
    - The zero regressors then stem from a distortion in the breathing amplitude histogram (the lower plot in the Preproc: Diagnostics figure does not contain a very informative histogram because of that) - The high oscillations dominate the histogram and lead to zero dynamic range of the respiratory phases when the regressors are flat, and a very small dynamic range when it "works" - note the spiky appearance of the respiratory RETROICOR regressors even in your first example.
    - This problem can be alleviated by removing these large oscillatory parts from the time series, we would have to think a bit how to do this properly within PhysIO, but I want to wait until you have confirmed the right `StartTime` values, maybe they already resolve this second issue as well, if the oscillatory segments are not part of the recording window of interest.

[image]

I hope that helps, looking forward to your response,
Lars

lbanellis, 2022-07-25:
Dears Lars,

Thank you for your detailed response and apologies for my delay in responding (I have been away from work). 

Thank you for noticing that StartTime should be negative if using BIDS formatting. I discussed with my colleague who converted the physio files to BIDS and we have now adjusted the StartTime relative to the first volume scan start (so as you suggested, StartTime is now negative as physio recording started before the first scan volume). Also, to confirm, log_files.relative_start_acquisition is 0. 

Apologies for including a filtered example, that was a mistake of mine as I was exploring both not filtered and filtered options to see if respiratory regressors may be estimated in either case. Respiratory regressors are still not estimated with not filtered (i.e., raw) respiratory data. 

However, even for the raw data, the 'Raw Physiological Logfile Data' plot does demonstrate a strong oscillation at the end of the time series (like you stated): 

[image]

This isn't in the raw data I submit to tapas (see raw data plot below):

[image]

Therefore, I don't think removing the end strong oscillation part of the data will work as this isn't present in the raw.

data from 2 subjects & script below:
[PhysIO.zip](https://github.com/translationalneuromodeling/tapas/files/9180220/PhysIO.zip)

Thank you in advance for helping me with this.

Best wishes

Leah

mrikasper, 2022-08-10:
Dear Leah,

Thank you for confirming the sign of `StartTime` and indeed, `log_files.relative_start_acquisition` should then be zero. I can reproduce your example, and the cardiac curves (looks like a pule oximeter to me) are a bit noisy, but fine for RETROICOR, HRV etc. This is clear if you zoom in a bit into the diagnostics plot for the heartbeats, which show robust peak detection and no "spikes" in the estimated peak-to-peak interval (it's always close to 1s, i.e., 60 bpm)

[image]

Interestingly, at least for this preprocessed version of the cardiac trace in this figure, there seems to be a second slower frequency with maxima every 6 seconds that is additive to the 1s cardiac signal...maybe you have the respiration already recorded there (would be good to know more about what the cardiac channel is exactly for your data)?

For the respiratory data, there seems to be indeed a bug in PhysIO's preprocessing, because your raw data does not have this spike at the end. I will get back to you about this until the end of the week.

All the best,
Lars

mrikasper, 2022-08-10:
After a bit of detective work, I found a possible reason:

In `tapas_physio_preprocess_phys_timeseries` for `sub-0019`, PhysIO finds that the time vector of the physio logging is only 838.6s long, but the scan duration 840s. So there is a discrepancy of 1 TR, and because the physiological trace is deemed too short, PhysIO tries to fill it up with zeros, which in your case creates a big step (because zero was not in the dynamic range of your signal).

I can definitely fix the arbitrary filling with 0 (I could use the max or mean), but this discrepancy is suspicious...could it be that your logfile is exactly 1 TR too short (maybe a trigger is ignored)? It looks a bit like that, because if I set `verbose.level = 3`, the raw BIDS data is plotted before the padding and looks like that:

[image]

There are exactly 600 volume start events (cyan, "volpulse") in your BIDS file, but as the graphics show, the last one is right at the end of the logging (and the first one at the very beginning). Typically, one would expect another TR of logging after the last scan trigger signal. So either the conversion is wrong, the timing of the logging, or you only acquired 599 volumes.

A quickfix to look at the respiratory regressors would be to set `Nscans=599` (see screenshots, you can see the modulation of the cardiac signal with the resp signal here as well), but I think you should rather follow up with this trigger mystery.

[image]

[image]

I hope that helps.

All the best,
Lars