translationalneuromodeling/tapas #202: How to apply BrainProducts data to PhysIO Toolbox

amijas, 2022-07-11:
Hi, everyone.
Thank you for the great work.

I got cardiac data and respiration data using BrainProducts, and I removed MRI noise from these data using EEGLAB(referring this site: https://fsl.fmrib.ox.ac.uk/eeglab/fmribplugin/).
So, I got the original cardiac and respiration data("data.eeg") and MRI-noise-removed data("data.CLEAN.set").

I'm in trouble because I don't understand which data I should use on SPM PhysIO Toolbox batch editer.
For "log_cardiac" and "log_respiration" items, I should put "data.eeg", "data.CLEAN.set", or anything else?

I'd appreciate it if you could answer my question.

[image]

mrikasper, 2022-07-13:
Dear @amijas,

Thank you for using PhysIO! I must confess it's been a while since we worked with BrainProducts data with some collaborators in Munich, so the reader might not be fully up to date. From what I can see in the [code](https://github.com/translationalneuromodeling/tapas/blob/master/PhysIO/code/readin/tapas_physio_read_physlogfiles_brainproducts.m), 

1. you will have to input the `.eeg` file, and then PhysIO relies on the fieldtrip version shipped with SPM to extract the data. 
2. There are some hardcoded assumptions, e.g., the scanner volume trigger being labeled `R128` that you might have to adjust.     
    - The selected channel by default is the filtered `ecg1`.
3. If the data cleaned by FSL can be read in by fieldtrip, you should be able to use it with PhysIO, maybe you have to adjust the channel label.
4.  The released code in PhysIO only supports the ECG channel data; for the respiration, our collaborators exported the trace into a text file.

If you can send me some example files and pointers which channel is the respiration channel, I could have a look.

All the best,
Lars

amijas, 2022-07-22:
Dear @Lars

Thank you for your reply. Your advice is really helpful and I can finally use PhysIO without error messages. 

Below is for people who plan to use PhysIO using Brain product data.
When you use brain-product-cardiac-and-respiratory `.eeg` data, you have to denoise these using EEG lab. When you've done denoise, you will obtain `.set` data. 
Then, you have to create `.txt` files that contain cardiac & resp signal data between R128s using MATLAB. 

Many thanks,
Amijas

closed 2022-07-31 by amijas