translationalneuromodeling/tapas #205: Error due to deprecated nanmean function

likeajumprope, 2022-07-25:
The nanmean function used in tapas_physio_filter_respiratory.m is deprecated (https://au.mathworks.com/help/stats/nanmean.html) and throws an error with Matlab R2022a when running examples.

mrikasper, 2022-07-25:
Dear Johanna,

it's not really deprecated (just not recommended :-)), but unlike `mean` requires the Matlab `stats` toolbox.

I have sketched a solution here: #206 , but paste it here for reference:

> You can replace `nanmean(X)` by `mean(X, 'omitnan)` or `nanmean(X,2)` by `mean(X,2,'omitnan')` for all occurences of nanmean, because mean should ship with any Matlab version.

However, there may be more functions relying on `stats` within PhysIO, so we have to carefully check that.

All the best,
Lars

mrikasper, 2022-07-25:
This command will give you an overview on the files and products (Matlab Toolboxes) PhysIO uses:

```
 [fList,pList] = matlab.codetools.requiredFilesAndProducts('tapas_physio_main_create_regressors.m')
```

closed 2022-10-25 by mrikasper