translationalneuromodeling/tapas #206: Possible issues with log file

marni-23, 2022-07-25:
Hi everyone!

I have physiological data from a 3T Phillips scanner and I would like to use this toolbox to create regressors for fMRI analyses. It is my first time using PhysIO and I seem to be having issues reading in my physiological files. When I tried running the script within SPM I've received the following error messages:

23-Jul-2022 12:13:20 - Running 'TAPAS PhysIO Toolbox'
No cardiac R-peak (heartbeat) events provided

23-Jul-2022 12:13:30 - Failed  'TAPAS PhysIO Toolbox'
Undefined function 'nanmean' for input arguments of type 'double'.
In file "/Users/marni/Downloads/tapas-master/PhysIO/code/preproc/tapas_physio_filter_respiratory.m" (???), function "tapas_physio_filter_respiratory" at line 65.
In file "/Users/marni/Downloads/tapas-master/PhysIO/code/tapas_physio_main_create_regressors.m" (???), function "tapas_physio_main_create_regressors" at line 225.
In file "/Users/marni/Downloads/tapas-master/PhysIO/code/tapas_physio_cfg_matlabbatch.m" (???), function "run_physio" at line 1661.
The following modules did not run:
Failed: TAPAS PhysIO Toolbox

I have attached here one of my log files in case there's an issue with the data itself? I have run this on a few other participants and I receive the same errors. 

Any help understanding the issue would be really appreciated!

Many thanks,
Marni
[ScanPsaLog20220701124502.log.zip](https://github.com/translationalneuromodeling/tapas/files/9183437/ScanPsaLog20220701124502.log.zip)

mrikasper, 2022-07-25:
Dear Marni,

Thank you for trying out PhysIO. It seems that you version of Matlab does not contain the `nanmean` function. Here are two questions to further help with your issue:

1. Which release of Matlab are you using?
2. Have you installed the Matlab *Statistics and Machine Learning Toolbox* (`stats`)?

You can find out both pieces of information by typing

`ver` 

in the Matlab command window. It should look something like that:

```
>> ver
-----------------------------------------------------------------------------------------------------
MATLAB Version: 9.7.0.1737446 (R2019b) Update 9
MATLAB License Number: 920805
Operating System: Microsoft Windows 10 Pro Version 10.0 (Build 19044)
Java Version: Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 9.7         (R2019b)   
Simulink                                              Version 10.0        (R2019b)   
Curve Fitting Toolbox                                 Version 3.5.10      (R2019b)   
DSP System Toolbox                                    Version 9.9         (R2019b)   
Data Acquisition Toolbox                              Version 4.0.1       (R2019b)   
Database Toolbox                                      Version 9.2         (R2019b)   
DatapixxToolbox                                       Version 0.9,        Aug        
Deep Learning Toolbox                                 Version 13.0        (R2019b)   
Filter Design HDL Coder                               Version 3.1.6       (R2019b)   
Fixed-Point Designer                                  Version 6.4         (R2019b)   
Global Optimization Toolbox                           Version 4.2         (R2019b)   
Image Processing Toolbox                              Version 11.0        (R2019b)   
MATLAB Compiler                                       Version 7.1         (R2019b)   
MATLAB Compiler SDK                                   Version 6.7         (R2019b)   
Optimization Toolbox                                  Version 8.4         (R2019b)   
Parallel Computing Toolbox                            Version 7.1         (R2019b)   
Psychtoolbox                                          Version 3.0.14      30 December
Reinforcement Learning Toolbox                        Version 1.1         (R2019b)   
Signal Processing Toolbox                             Version 8.3         (R2019b)   
Statistical Parametric Mapping                        Version 7771        (SPM12)    
Statistics and Machine Learning Toolbox               Version 11.6        (R2019b)   
Symbolic Math Toolbox                                 Version 8.4         (R2019b)   
Wavelet Toolbox                                       Version 5.3         (R2019b)   
```

If you haven't got the statistics toolbox, the code won't run. There are several solutions:

1. You can just install the Statistics Toolbox with the Matlab installer, if your institution has the license.
2. You can use a [precompiled version of PhysIO within NeuroDesk](https://gitlab.ethz.ch/physio/physio-doc/-/wikis/FAQ#15-i-dont-have-matlab-can-i-still-use-the-physio-toolbox) that doesn't need the license
3. You can replace `nanmean(X)` by `mean(X, 'omitnan`) or `nanmean(X,2)` by `mean(X,2,'omitnan')` for all occurences of `nanmean`, because `mean` should ship with any Matlab version. @likeajumprope is working on a bugfix for PhysIO along this road, which will probably be released in September, but the substitution in your local file should be very straightforward.
    - See #205 
    - There might still be other functions within the Statistics Toolbox that PhysIO uses (we will check), so this might not fully resolve the issue.

Please let me know how you proceed.

All the best,
Lars

marni-23, 2022-07-25:
Hi Lars, 

Thanks for getting back to me so quickly! 

Regarding your questions, I'm running on MATLAB Version: 9.9.0.1718557 (R2020b). It seems that I had not installed the Statistics Toolbox - that seems to have fixed this issue. Thanks! 

Best wishes,
Marni

closed 2022-07-29 by mrikasper