translationalneuromodeling/tapas #211: tapas_physio_check_spm_batch_editor_integration should honour spm_get_defaults('tbx') for possible toolbox installation paths

gdevenyi, 2022-09-06:
When initializing PhysIO, tapas checks a hardcoded path:
https://github.com/translationalneuromodeling/tapas/blob/master/PhysIO/code/utils/tapas_physio_check_spm_batch_editor_integration.m#L40

But I have separately managed toolbox directories, by adding this file to the startup path:
```
$ cat spm_my_defaults.m 
global defaults
defaults.tbx.dir{end+1} = fullfile(fileparts(mfilename('fullpath')),"PhysIO","toolbox") ;
```

Where `toolbox` contains the linked `code/` directory, named `PhysIO`

This is to enable `lmod` (https://lmod.readthedocs.io/en/latest/) to (potentially) manage multiple parallel versions of toolboxes for users.

mrikasper, 2022-09-07:
Dear Gabriel,

Thank you for the feedback and pointing us to the `lmod` system. How would I go about this? Checking every directory in the `defaults.tbx.dir` cell array and its subdirectory?

Just out of curiosity: Do you have an example of using different versions of SPM toolboxes with `lmod`?

All the best,
Lars

gdevenyi, 2022-09-08:
> How would I go about this? Checking every directory in the defaults.tbx.dir cell array and its subdirectories?

Yes I think that would resolve this particular check for my implementation.

> Just out of curiosity: Do you have an example of using different versions of SPM toolboxes with lmod?

The spm startup code will find and load any `spm_my_defaults.m` file it finds on the MATLABPATH:
https://github.com/neurodebian/spm12/blob/master/spm_defaults.m#L200-L236

So my lmod modules consist of:
`prepend_path( "MATLABPATH",          "/path/to/tapas/<VERSION>/install")`

Which then contains a `spm_my_defaults.m` with matlab code I provided above, appending a dir to the toolbox path which contains the toolbox.

If I want different versions, I just create a different `/path/to/tapas/<VERSION>/install` and place that version there.

For full details I have the skeleton of my lmod system here,
https://github.com/DouglasNeuroInformatics/software_quarantine/blob/master/software/tapas/6.0.1/module

There's no examples for multiple versions yet explicitly because this is a fresh rebuild transitioning from environment-modules (TCL) to lmod (lua)

mrikasper, 2022-09-09:
Dear Gabriel, 

That's great to hear. I have changed `tapas_physio_check_spm_batch_editor_integration` in the [physio branch](https://github.com/translationalneuromodeling/tapas/tree/physio) following your feedback. Could you please check whether it fulfills your needs? If we are quick, it might make it into the upcoming release.

All the best,
Lars

gdevenyi, 2022-09-09:
`tapas_init` now works without any code modifications:
```
Checking Matlab PhysIO paths, SPM paths and Batch Editor integration now...
Checking Matlab PhysIO paths now...added PhysIO path recursively: /tmp/tapas/PhysIO/code
Checking Matlab SPM path now...OK.
Checking whether PhysIO/code folder is a subfolder of SPM/toolbox (or a link within there)...Defaults settings have been modified by file(s):
  /tmp/tapas/spm_my_defaults.m
Modified fields: tbx 
```

One small thing, there's a missing newline after the "..."