translationalneuromodeling/tapas #216: regressor text file on FSL

Ruwanmri, 2022-11-11:
Hi,

Can anyone advise me please how to integrate multiple regressor text files for the first-level analysis in FSL? 
Is it correct if we use the text file in  'add additional confound EVs'?

Thank you.

Best regards,
Ruwan

mrikasper, 2022-11-11:
Dear Ruwan,

Thank you for using the PhysIO Toolbox - at least I am assuming you mean the `multiple_regressors.txt` files created by it?

You are right, if I am reading the [FSL FEAT](https://fsl.fmrib.ox.ac.uk/fsl/fslwiki/FEAT/UserGuide) documentation correctly, you should just be able to add the text file as you describe. Quoting the FSL manual here (with my highlights):

> If you want to add other confound EVs than motion parameters, that you have precomputed for your data, then turn the **Add additional confound EVs** option on and then enter the filename of a raw text file (or, if you are setting up multiple analyses, enter one text file for each input FMRI dataset to be processed). The _file can contain as many confound EVs as you like, each in a separate column in the text file_.

This is exactly how the text files are organized by PhysIO as well.

All the best,
Lars

Ruwanmri, 2022-11-11:
Hi Lars,

Thank you very much. Yes I meant the multiple_regressors.txt file resulting from PhysIO. 
Indeed I also went through this part and tried it there. It looked fine too. I wanted to make sure that I am doing the right thing with it. Thank you very much for confirming that.

Best regards,
Ruwan

closed 2022-11-11 by Ruwanmri