translationalneuromodeling/tapas #222: hgf_ar1_mab: priors in unstable region for this startpoint.

ainuura, 2022-12-19:
Dear HGF experts,

I am trying to develop an multi-armed bandit experiment, where participants have to choose one out of three bandits at each trial, with the bandits having varying payout magnitudes over time. However, I am struggling to use the payouts as input using the (2-level) hgf_ar1_mab model. Specifically, I am running into this error message when trying fit the model using `tapas_fitModel(y, [ u y ], 'tapas_hgf_ar1_mab_config', 'tapas_softmax_config');` with y being a column vector with elements { 1,2,3 } coding for the selected bandit and u is the payout (between 1 and 100) retrieved at trial t, respectively:

```
Warning: priors in unstable region for this startpoint.
Re-sampling startpoints...
Warning: priors in unstable region for this startpoint.
Re-sampling startpoints...
.
.
Model inversion aborted. No stable startpoint found for the current priors in 1000 startpoint sampling iterations.
```

I tried tweaking the priors (making them wider, fixing alpa), but I just can't seem to make the priors appropriate for the 2-level model. 
Right now I am really wondering if this is a problem with my inputs or if I miss something important while setting the priors in the config file. When I simplify the input to binary (e.g. "high pay"/"low pay"), I get sensible results using the `tapas_hgf_ar1_binary_mab` model. I would be grateful if someone could point me into the right direction.

Best,
Ananda

chmathys, 2022-12-20:
Hi Ananda,

It seem like you have a setup much like the one in this paper:

Reed, E. J., Uddenberg, S., Suthaharan, P., Mathys, C. D., Taylor, J. R., Groman, S. M., & Corlett, P. R. (2020). Paranoia as a deficit in non-social belief updating. ELife, 9, e56345. https://doi.org/10.7554/eLife.56345

Is that correct? If yes, you might try comparing how exactly your situation is different. This could help you find out in what way to adjust your priors.

It may also help to narrow your priors instead of widening them.

ainuura, 2022-12-20:
Dear Dr. Mathys,

Unfortunately, tightening the priors did not improve the situation either. I think the most comparable setup is the one used in Daw et al. (2006, https://www.nature.com/articles/nature04766), only having three instead of four bandits. The mean payout magnitude changes after some time for each bandit, and, additionally, we add some random noise to the payout magnitude. So I would have assumed that the input u corresponds to the payout magnitude at trial t, and y(t) to the selected bandit. Is there something else I could try to find suitable priors?

Thank you very much for your help!

chmathys, 2022-12-20:
That all seems correct. You could try turning the problem around: choose a set of parameters that seem plausible and simulate data with these, if they work. If they don't, change them until you've got one setting that works and work from that. Once you've got a working setting, explore how the simulated data change with changes in the parameters. That'll give you an idea of the plausible ranges for the parameters.

ahmadkhanloo, 2024-01-20:
Hi @ainuura 

Could you share your final solution?