translationalneuromodeling/tapas #223: Is it necessary to separate by run?

Kwon-13338, 2023-01-04:
Dear experts.

I just trying to apply physiO toolbox for biopac.txt data. 
The biopack data includes the entire fMRI scan process, and in this process, all three runs are included in one file. 
For fMRI conditions, tr was 2, slicetiming was 90 in one run, and scan was executed 365 times per run. 
Do I have to manually edit and apply the biopack data in three runs? 
Does the toolbox recognize three runs and give the full regressor? 
If I want to get the regressor for each of the three runs, I need advice on how to do it. 
Also, I am not sure how to set the parameters of the detailed physiO toolbox.
I have attached the regressor result, code, and biopac data entered as input.

Best,
yonghyun. 
[image]
[image]
[20220801_DP01_Andy.zip](https://github.com/translationalneuromodeling/tapas/files/10344784/20220801_DP01_Andy.zip)

mrikasper, 2023-01-24:
Dear Yonghyun,

My apologies for the late reply, I am on parental leave now, so things might be a bit slow in general on my end.

I haven't had time to run your code yet, but could you please send me a `.m` file of the batch job instead of a `.png` screenshot?

In general, PhysIO should be able to just extract the part of the logfile for each run that you need. Basically, you would specify a different temporal offset for each run and leave all other parameters the same. There are two principal ways of doing so:

1. Define a different `logfiles.relative_start_acquisition` for each run, with an offset that is roughly `Nscans*TR` (Nscans is the number of volumes per run). This would need a bit of trial and error adjustment looking at the raw time series plots that PhysIO provides (esp. for `verbose.level = 3`).
2. If the BioPac file has triggers per volume, you could just alter the number of "dummy scans" in `scan_timing.sqpar.Ndummies` to reflect the accumulated number of triggers sent in each run.

In both cases, you should set `log_files.align_scan = 'first'`, so that volumes and time are counted from the beginning of the log file.

I will have a look whether I can create an example once you sent me the `.m` batch file. Thank you!

All the best,
Lars

Kwon-13338, 2023-01-25:
Dear Lars,

Thank you so much for taking the time to respond. As you said, I will try to apply physIO again.
However, while you are answering, I have a few questions.

1. What does a dummy scan mean? As I said, 1131 triggers were confirmed, including 365 scans for each of the three Runs, 30 pre-rest scans, and two reverse scans after each run. If so, are the rest of the scans other than run put into Ndummies separately from Nscans?

2. When I looked up the other issues, I found one that recommended setting the middle slice as onset.slice to sample from the number of slices. So I set 45, which is half the number of slices per volume. Is this the right choice?

3. After GLM the regressor output as a result, tSNR verification was attempted for the residual. I'm not a SPM user, so I used a tool I'm more familiar with, AFNI, for tSNR. However, there was a big difference between the tSNR (threshold range: 0 to 232) of the subjects who had only undergone preprocessing and the tSNR (threshold range: 0 to 3) of the residuals after the physIO regressor and GLM (the result itself looks strange). Any advice on this matter

As you said, I have attached the .m file and the biopac file for input.
I did not understand the application of offset that additionally advised. Is it possible to set per run-through parameter without modifying the input file? Also, if there is a parameter set incorrectly, please advise.
If it's too intrusive during the holiday period, we'd appreciate it if you could leave a comment on the question.

Thank you.
Yonghyun.
[yonghyun.zip](https://github.com/translationalneuromodeling/tapas/files/10502747/yonghyun.zip)