translationalneuromodeling/tapas #225: tapas_physio_read_physlogfiles_brainproducts.m function unusable

ChristophWittkamp, 2023-02-01:
Basically there is no input for the output Variables (c, r, t, cpulse, acq_codes), there is no variable "file_path" and no variable "file_name" when it comes to saving and the ft_triafun_segment function used does not exist.

mrikasper, 2023-02-02:
Dear Christoph,

Thank you for trying out PhysIO and my apologies it has been such a disappointing experience.

The BrainProducts reader was developed in a collaboration a couple of years back and worked for a specific study, without being fully integrated into the toolbox interface. It could be that some functions of the original code are therefore missing.

Also, I couldn't include the example files for that use case at the time. Thus, I did not devise any unit and integration tests and with interface changes of the toolbox over the years, this code has probably deteriorated into the state you now find it in.

If you are interested in reviving it, maybe you could explain your use case a bit or send some example files, so I understand better what you are trying to do.

All the best,
Lars

ChristophWittkamp, 2023-02-21:
Dear Lars,

Thank you for your really usefull toolbox. I really appreciate your work. I now used the 'Biopac_Mat' Vendor format and it works fine. For anyone interested it is even possible to implement R-Peaks that our workgroup already extracted in the BrainVision Analyzer. 

Thanks again and all the best,
Christoph

mrikasper, 2023-02-23:
Dear Christoph,

I am glad to hear this export route worked! Did you somehow convert the BrainProducts format to BioPac or did you use different data that you had available? Just curious whether this could be a generic route for the BrainProducts data usage in PhysIO.

All the best,
Lars

ChristophWittkamp, 2023-02-23:
Sadly we just used the BrainVision Analyzer for R-Peak-Detection (which we used for Artifact correction in EEG-data anyway.) The ECG data i used is recorded with a Biopac device (but was not stored in the correct format).

Kind regards,
Christoph

mrikasper, 2023-02-23:
Dear Christoph,

Thank you for clarifying. I will close this issue for now, but am happy to answer any further questions regarding your data.

Good luck with the analysis!

All the best,
Lars

closed 2023-02-23 by mrikasper