translationalneuromodeling/tapas #226: rDCM in Python

jadecci, 2023-02-07:
Hi :)

I am interested in using rDCM in a behaviour prediction project currently in preparation phase. As my code are all in Python and in a nipype workflow, I would like to compute rDCM in the Python framework. May I know how I should proceed with this?

If there is no plan for Python release of the rDCM toolbox, would it be cool if I translate some of the code to Python and submit it to the main repository here?

Thank you!

StefanFraessle, 2023-02-24:
Hi @jadecci 

Thank you so much for your interest in rDCM and I hope that it will prove useful for you. Also, my apologies for the delay in responding - I'm currently in between two jobs and the transition period is a bit busy.

You are correct that there is currently no Python version of the rDCM toolbox - and there are no concrete plans to translate rDCM into Python at the moment. I see several steps forward:

- If you do have access to MATLAB (as a student, you should get it for free via your university) then you could simply perform all of your analyses until the BOLD signal time series within Python (or whatever language you want to use), then take the extracted time series into MATLAB and use the original rDCM toolbox. The inferred connectivity parameter estimates from rDCM could then again be loaded into Python for further analyses. While this might be a bit cumbersome, it would do the trick and should be rather straightforward given Python's capacity to read and write *.mat files.
- We are currently working on a Julia version of the rDCM toolbox. If you are anyway working with Julia already, then this might be a good alternative as well. However, it is a bit difficult to predict at the moment when exactly that Julia version will be released.
- Finally, you could write your own Python version of the rDCM code. The license under which the rDCM toolbox (and, in fact, the entire TAPAS software package) is published allows you to do so. Having said this, I feel that this is likely considerably more work than - for instance - option 1.

If you decide for the latter option and are interested in submitting your Python code to the TAPAS main repository, this is probably something that would need to be discussed at the given time. While external people have contributed small bits to existing toolboxes, we usually make sure that more "substantial" contributions to TAPAS are made by people that are affiliated with the TNU. Hence, whether or not a Python-rDCM should go into the main TAPAS repository would be a strategic decision that probably needs to be discussed with our professor.

Hope this is helpful already. If not, please feel free to come back to us.

All the best
Stefan

jadecci, 2023-05-08:
Dear @StefanFraessle 

Thanks for the help! I have written my own Python version of the code for the 'original' rDCM model (https://github.com/jadecci/rDCM_py). It replicates the Matlab code with a very small margin of error and is working well for my need :)

If you think that this could be a useful addition to the TAPAS main repository, I would be happy to add the other functionalities as well (e.g. sparse model).