translationalneuromodeling/tapas #227: Long form physio trace from Siemens machine

HeardMatthew, 2023-02-17:
Hello, 

Thanks so much for maintaining such an excellent toolbox, I think this project is the perfect solution for in my current dataset. I wanted to verify that this toolbox can handle my current obstacle, or if I need to process my data before feeding it into the toolbox. 

I have a physio trace (breathing belt and pulse oximeter) collected from a Siemens Prisma 3T and saved using the VB (.resp and .puls) file format. The scanning technician started physio recording as soon as the participant was put inside the scanner, and let the script run for the duration of the entire scan (~30 minutes). In particular, I would like to create a physiological regressor for a resting state scan acquired about 10 minutes into the scan, and lasted for 10 minutes. 

Can I feed the "long form", raw physio trace into the toolbox and instruct it to create a physio regressor for the specific scan? Or do I need to break the physio trace into smaller chunks? 

Thanks!
Matthew

mrikasper, 2023-02-17:
Dear Matthew,

Thank you for your positive feedback regarding PhysIO!

I believe you can (and should) use the "long form" trace out of the box. It might take PhysIO a bit longer to process a larger file, but splitting it up the Siemens VB format is not so easy, because the trace is "sandwiched" between a header and a footer part in the files (See [here](https://gitlab.ethz.ch/physio/physio-doc/-/wikis/MANUAL_PART_READIN#manual-recording) for a description).

But what you do have to get right is the temporal synchronization of your 10 min resting state scan within this long trace. There are basically two options here

1. You have a DICOM (header) file for the first volume of your resting-state scan. In this case, you can just set it as the `log_files.scan_timing` file and `log_files.align_scan = 'first'`, so PhysIO will match the timestamps of log file and DICOM, similar to what I describe [here](https://github.com/translationalneuromodeling/tapas/issues/218#issuecomment-1402136995) for a slightly different issue.
2. You have an external TTL trigger from the scanner connected to your physiological recording box, so the `.ext` files are populated with trigger traces (instead of all 0, you can check the file content with a text editor). I am working on a solution to support these files as `log_files.scan_timing` input as well (inspired by [this issue](https://github.com/translationalneuromodeling/tapas/issues/201) that shows the DICOM header solution is not always reliable, in particular for very long trace files). If you also have a need for this, it will bubble up my priority list of new feature implementations.

All the best,
Lars

mrikasper, 2023-02-23:
Dear Matthew,

I agree with your choice for synchronization. Regarding using the `first` or `last` scan volume for synchronization, in your case (Siemens) it's fine to use either, as long as you know which DICOM file pertains to the first and last volume you want to use in your fMRI analysis (Sometimes people prefer the first one, because then they have to change less in their scripts, because the number of volumes might vary between subjects or runs).

The general recommendation for the last scan volume comes from the fact that on some scanner vendors, it's not always clear whether certain scan preparation modules (shimming/frequency adjustment) or dummy scans (to facilitate T1 equilibration) elicit a trigger signal, so you might end up with a wrong synchronization using the `first` trigger. 

On the other hand, the only time I wouldn't recommend using the last volume is if you manually stop the sequence and there could be incomplete volumes (only a few slices) at the end of a run and the related logfile. But again, not an issue on Siemens usually, because you manually start/stop the logging.

I hope that clarifies the issue.

All the best,
Lars

zhengchencai, 2025-02-22:
Dear Lars,

Thanks for developing this tool and explaining the details for syncing physio data with fMRI. I also encountered the same [which clock to use issue](https://github.com/translationalneuromodeling/tapas/issues/201). In my case, the MDH starts from 1 am, which does not make any sense, but the MPCU start really matches the AcquisitionTime tag in the BIDS JSON as well as the real-life acquisition time (3pm).

We also got .ext file that contain 3 unique values (apart from the first 4 header values): 0, 1, and 5000. These 5000 values match very precisely with the fMRI runs by the number of TRs. One consistent observation across different scans is that the number of data points in the .puls, .ext, and .resp files are exactly the same (without removing 5000 after removing headers). So I guess using 5000 in .ext can sync physio data with fMRI more confidently, actually, also less computations. 

Another interesting observation is that the 5000 values in each of the 3 files are not the same; they appear as a consistent trigger in the .puls and .resp files with some miss samples sometimes, which does not follow the fMRI TR at all. I assume these are the triggers from the PMU to the sensor (see figure below). The puls channel has much more frequent 5000s than resp. Also, we have exactly the same stop time in the two clocks across the 3 files, but the start time differs a bit...

[image]
here we did 50 TRs in one short testing run, the unique(diff(index of 5000)) are 801 (49/50 TRs), 800 (the last) and (1/50 TRs). (801-1)*2.5ms/1000 = 2s which is our TR.

[image]
same chunk in resp

[image]
same chunk in puls

After seeing this, I am not sure how to handle the 5000 values. Most of the content I read suggests that 5000 should be removed, but there is no clear explanation regarding whether 5000 occupies a time tick. If it does, that means we should interpolate the values after removing 5000 in resp and puls. If not, I have a hard time understanding how the machine can insert 5000 for 3 different channels and keep the remaining values precisely at 400 Hz. Do you know more details or have a more firm source of information?

Thank you very much.

mrikasper, 2025-02-24:
Dear @zhengchencai,

Thank you for looking into the trigger issue in such a detailed way. Here are my thoughts:

1. It makes sense for you to use the `MPCU` clock. PhysIO will soon move into a [new repository](https://github.com/ComputationalPsychiatry/PhysIO), and you can already download the current development version there. Then, you can change the code in [this line](https://github.com/ComputationalPsychiatry/PhysIO/blob/94d5fc22cf424e1739b11d0da3a98e86c5748927/code/readin/tapas_physio_read_physlogfiles_siemens.m#L164) to `referenceClockString = 'MPCU'; `
2. I agree, the `5000` markers in your `.ext` files are a better way to synchronize your data. Probably your scanner hardware is set up in a nice way, so that the scanner trigger signal is fed back into the `.ext` channel (which is not the default). I started enabling the use of `.ext` files as `scan_timing_log` file in the mentioned development version, please try it out and provide feedback whether it works. PhysIO takes care of the header in these files automatically and respects the mentioned start and end times, so I don't think you have to modify the files in any way and can hand them to PhysIO as is.
3. The `5000` markers in the `.resp` and `.puls` files indicate something else, i.e., the physiological trigger times. If you had a gated/triggered sequence, these markers would indicate when the scanner had detected a maximum in- (or ex-)halation point in the trace, or a maximum in the pulse trace, respectively. PhysIO does this peak detection automatically using its own robust method, so you don't have to use the `5000` from these files (PhysIO should just ignore them when reading the file, let me know if it does not work).

In summary, PhysIO should be able to use your files and synchronize them correctly either by specifying `MPCU` as the clock (and not specifying a `scan_timing_log` file) or by specifying `.ext` as the `scan_timing_log` file, with the latter method presumably being more accurate.

I hope that helps!

All the best,
Lars

zhengchencai, 2025-02-24:
Hi @mrikasper, thank you very much for you detailed explanation. I will try the new repository for sure. 

I am still unclear about how the value 5000 is handled in the final processing. In this [line](https://github.com/ComputationalPsychiatry/PhysIO/blob/94d5fc22cf424e1739b11d0da3a98e86c5748927/code/readin/tapas_physio_siemens_line2table.m#L172), the indices of 5000 are stored. Are they used later in the pipeline?

I checked the number of data points in the .ext, .puls, and .resp files, and **they are identical after removing the headers while keeping 5000s**. Using the position of 5000 in .ext, I can align it with the fMRI run in .puls and .resp. However, should we interpolate values at these positions in .puls and .resp?

As shown in the figure above, the number of 5000s differs significantly across the three files (it is only one short test fMRI run with 50 volumes, in a full scan for 1h, the amount of 5000 in three files are 735, 777 and 3406 in resp, ext and puls, respectively). If I simply remove 5000 without assuming it occupies a time tick (e.g., for the sequence 856 5000 678, treating the time delay from 856 to 678 as 2.5 ms instead of 5 ms), the .resp and .puls files will have different durations. Since they contain different numbers of 5000s, this discrepancy could introduce significant errors when aligning physiological data to fMRI, regardless of whether the log method or timestamp method is used. As you mentioned, if 5000 are the peaks of resp and pulse, shouldn't we interpolate the values after removing 5000? Sorry for insisting on this 5000, I just want to understand the details better. Most of discussions I saw are removing 5000, people usually do data[data < 4096], compute the scan duration and then assign time stamp to data assuming it is a consistent 2.5ms interval. In AFNI, 5000 are interpolated in all examples of [physio_calc](https://afni.nimh.nih.gov/pub/dist/doc/program_help/physio_calc.py.html), but the documentation also says we should remove them rather than interpolate... 

I also tried to remove 5000s and them compute the total scan time by len(data[data<4096]) * 2.5, hoping it should match with the duration (stoptime - starttime) in all three files, but it is not the case...

After reading your text 

> the physiological trigger times. If you had a gated/triggered sequence, these markers would indicate when the scanner had detected a maximum in- (or ex-)halation point in the trace, or a maximum in [...]

do you think this implies that the scanner is detecting the peaks itself to estimate the pulse and respiration frequency reported in the footer? In other words, it uses the 5000 values to derive a rough frequency for the summary report. I counted the number of 5000s in the respiration and pulse signals after segmenting them into fMRI runs based on the 5000s in the .ext file. Then, I computed the frequency of 5000 occurrences in the respiration and pulse signals, and they matched the heart rate and respiration frequency. For instance, in a 100s respiration chunk, there were 30 occurrences of 5000, while in a pulse chunk, there were 134 occurrences (I tend to be quite nervous inside the scanner...). If this is the case, it seems we should interpolate after removing the 5000s.

mrikasper, 2025-02-26:
Dear @zhengchencai, 

Thank you for diving into this further!

Please see my responses below:

> I am still unclear about how the value 5000 is handled in the final processing. In this [...]
The physiological triggers are used as detected peaks, if you set `preproc.cardiac.initial_cpulse_select.method = 'load_from_logfile';` instead of the `auto_matched` default peak detection algorithm. From experience, the auto-detection is more robust than the `5000` in the physiological logfiles, because their detection is necessarily prospective (i.e., cannot use future patterns in the traces for peak detection).

> I checked the number of data points in the .ext, .puls, and .resp files, and **they are identical after removing the headers while keeping 5000s**. Using the position of 5000 in .ext, I can align it [...]

I am not sure I fully understand. I am surprised that the numbers are exactly the same while keeping the 5000s, because this would mean that the system somehow puts dummy values in a file where only the other physiological trace has a trigger event. Usually, the files also have tags in them that indicate, for example, changes in the gain of the recording device (e.g., `currentCushionGain: 25`) that would depend on the actual measurement. Which Siemens logfile version is indicated at the top of the header(e.g, `LOGVERSION_RESP   3`)?  I think it would be easiest if you could send me a set of these files to understand better.

> As shown in the figure above, the number of 5000s differs significantly across the three files (it is only one short test fMRI run with 50 volumes, in a full scan for 1h, the amount of 5000 in three [...]

I think the 5000s should just be removed and don't occupy an extra time step of 2.5ms - this is what you mean by "interpolating", correct? My main justification for this assumption is that I would not know what the system does for issues like the tags above that occupy 2 words (`currentCushionGain` and `25`). Also, the ECG recordings (and modern breathing `.resp` files with multiple sensors integrated in the bed) have multiple channels that should all be measured at the same time. It would be a very bad design if each channel is only sampled every `nChannels * 2.5ms`. It could be possible, but a priori I don't find it very likely.

> 
> I also tried to remove 5000s and them compute the total scan time by len(data[data<4096]) * 2.5, hoping it should match with the duration (stoptime - starttime) in all three files, but it is not the case...

OK, but with the 5000s included, it matches?

> do you think this implies that the scanner is detecting the peaks itself to estimate the pulse and respiration frequency reported in the footer? In other words, it uses the 5000 values to derive a rough frequency for the summary report. 

Yes, that's exactly what the scanner does. You can see it in the live display while scanning that it highlights these peaks.

> I counted the number of 5000s in the respiration and pulse signals after segmenting them into fMRI runs based on the 5000s in the .ext file. Then, I computed the frequency of 5000 occurrences in the [...]

I am not sure that your finding supports that interpolating (i.e., adding an extra 2.5ms to the timecourse) is better than removing. `30*2.5ms=75ms`, that will not change the estimated respiration rate much.

I hope that helps.

All the best,
Lars

zhengchencai, 2025-02-26:
Dear @mrikasper,

Thank you very much for your detailed and prompt reply and for your patience. I truly appreciate it.

Sure, please find the files [here](https://drive.google.com/drive/folders/1kOLeRto2zRkYOglNNxp8-wLUqkNYDV95?usp=sharing). They are acquired on Siemens Terra 7T using syngo MR E12. There are multiple short fMRI test runs, with TR of 2s (50 vol each run) or 1.5s (67 vol each run).

>I am not sure I fully understand. I am surprised that the numbers are exactly the same while keeping the 5000s, because this would mean that the system somehow puts dummy values in a file where only the other physiological trace has a trigger event.

Indeed the number of data samples are the same in 3 files after removing the header and footer **with 5000s**. Without 5000s, they do not match, there are much more 5000s in pulse than in other two. Therefore, if we remove 5000s, and assign 2.5ms for each data point, the time difference in the last data point across 3 files will be considerably different. If one does so after taking out the runs according to fMRI runs recorded in .ext, then the timing difference can be smaller. 

> I think the 5000s should just be removed and don't occupy an extra time step of 2.5ms - this is what you mean by "interpolating", correct? 

I mean interpolating by (1) removing all 5000 and (2) interpolating values at the positions of 5000s. One can either simply take the average of the value before and after each 5000 or do a spline interpolation. In other words, I assume each 5000 represents a single time point, at least based on my observations in our files.

> OK, but with the 5000s included, it matches?

No, nothing matches, if we stick with exact 400Hz sampling rate. It is rather around 399.9x, with/without 5000, but then we can not verify whether 5000 takes a time step or not by matching with the sampling rate estimated in this way.  

> I am not sure that your finding supports that interpolating (i.e., adding an extra 2.5ms to the timecourse) is better than removing. 30*2.5ms=75ms, that will not change the estimated respiration rate much.

I am not claiming which approach is better. I agree that removing them or not may not make a significant difference. Just want to understand how 5000 is recorded in the trace. A 75 ms interval is not that negligible because the correction is slice-based, and in our case, the slice interval is 80 ms.

I suspect this discrepancy is related to the recording method. In our case, the first number in the header is 1, which indicates method = None. However, I could not find out what this method actually does. In most cases shared by others, the method is set to 2, which corresponds to triggering.