translationalneuromodeling/tapas #229: Comparing groups using bayesian parameter average in HGF

milan-andrejevic, 2023-03-23:
Hi everyone, Hey Chris,

The HGF manual mentions that we can use tapas_bayesian_parameter_average() function to compare parameter averages across groups of subjects. I really appreciate that the function takes the parameter covariance structure and precision into account when estimating the means across subjects, but I am still not clear on how to use these point estimates to make group comparisons, test hypotheses, and make statistical inferences.

Is there an easy way to get credibility/confidence intervals of these estimates, or calculate Bayes Factors?

Thanks heaps!

Best wishes,
Milan

BPaton, 2023-03-23:
Hi Milan,

I have replied offline to you but some of the details of this are on pages
8 and 9 of the HGF manual.

Thanks.

Bryan.

milan-andrejevic, 2023-06-30:
Hi @chmathys (and @BPaton),

Bryan and I had a bit more thorough look at the issue, and concluded that the method of comparing log model evidence (LME) sums (currently described on pages 8 and 9 of the HGF manual as a way to do stats for BPA), does not quite do the job we try to compare averages across two experimental groups (between-subject design) and there are several free parameters in the model.

Bryan found [this page](https://www.visiondummy.com/2014/04/draw-error-ellipse-representing-covariance-matrix/) that describes how to calculate confidence intervals from covariance matrices, and suggested I could use the same method to calculate 95% credibility intervals of BPA group average estimates.

Here I am sharing the solution to provide a straightforward answer to anyone else trying to do compare parameter averages across groups using the BPA function.

`
% Run the BPA script on a group of model fits.
bpaGroup1 = tapas_bayesian_parameter_average(modelParticipant1, modelParticipant3, modelParticipant5);

% Calculate CIs for group 1
eigenValues = eig(bpaGroup1.optim.Sigma);
probCutOff = 0.95;
degreesFreedom = length(bpaGroup1.optim.Sigma);
chi2Percentile = chi2inv(probCutOff, degreesFreedom);
bpaGroup1CI = 2 * sqrt(chi2Percentile * eigenValues);
`
The idea is then to repeat the same procedure for the second group, and see if the upper credibility bound of the lower group mean estimate and the lower credibility bound of the higher group mean estimate overlap. If they don't we can claim that >95% chance that there's a difference.

It would be great if you can check it out and endorse it or let us know if you see any issues with the maths or the inferences.

Best wishes,
Milan