translationalneuromodeling/tapas #232: Landscape orientation for figures

NinetLab-Redux, 2023-04-30:
Hi Lars,
The figures are saved in portrait and some looked squished a bit to use in presentations. 
I tried to tinker with the "tapas_physio_print_figs_to_file.m"  and tried this on lines 70-72

   for k=1:length(verbose.fig_handles)
                set(verbose.fig_handles(k), 'PaperOrientation', 'landscape');
                print(verbose.fig_handles(k), printFormat,'-landscape', fullfile(pfx,[fn sprintf('_%02d', k) sfx]));

but it seems -landscape is an illegal option in Matlab 2020a.
I tried getting advice from the "Oracle" and suggested I use "print(verbose.fig_handles(k), printFormat, '-fillpage', fullfile(pfx,[fn sprintf('_%02d', k) sfx]));"
but I get
"Error using validate (line 135)
The 'fillpage' option is only valid for page formats and printers."
Any suggestions? I'm trying to create a flow with the images for Quality Assurance purposes
thanks and best regards

mrikasper, 2023-05-01:
Dear Fidel,

I believe the best way to go about this, is to change the `Position` of the figure, e.g., `set(verbose.fig_handles(k),'Position', [10 10 widthInPixel HeightInPixel]`. You might also adjust it's `WindowStyle` to `'default'`.

I hope that helps - let me know how your QA flow is going. There should also be an update this month to PhysIO with a much more versatile version of `tapas_physio_review` by @likeajumprope that should allow you to interactively recreate most of the figures from the `physio.mat` saved during processing.

All the best,
Lars

NinetLab-Redux, 2023-05-02:
Thanks, Lars!
I feel I don’t have basic Matlab understanding (trying to learn as I go)

I get this error “Unrecognized function or variable 'widthInPixel’. “
When I tried

for k=1:length(verbose.fig_handles)
                set(verbose.fig_handles(k),'Position', [10 10 widthInPixel HeightInPixel]);
                print(verbose.fig_handles(k),printFormat,fullfile(pfx,[fn sprintf('_%02d', k) sfx]));
            end

Maybe I’ll just wait until the end of the month for that update :)

fvr

[CAUTION: Non-UBC Email]

mrikasper, 2023-05-02:
Dear Fidel,

This was just a placeholder, you have to define values for these variables before, e.g., `widthInPixel = 1400; heightInPixel = 875;`.

All the best,
Lars

NinetLab-Redux, 2023-05-02:
Thanks, Lars!
I gathered it must had to do with my not knowing much about Matlab :)
I’ll give this a go!
best,
fvr