translationalneuromodeling/tapas #235: Problems defining trials

EvelyneFraats, 2023-06-05:
Hi all. I am trying to make cardiac and respiratory regressors using the PhysIO toolbox in SPM. I have one .EEG file, including my respiratory and cardiac measures. I also have one .vhdr and .mkr file.  When I run the code, I get the following error, indicating a problem with reading in the events. How can I change event_type or FT_DEFINETRIAL?

`05-Jun-2023 12:40:50 - Running job #1
------------------------------------------------------------------------
05-Jun-2023 12:40:50 - Running 'TAPAS PhysIO Toolbox'
Warning: cfg.trialdef.eventtype='?' is deprecated, please specify cfg.trialfun='ft_trialfun_show' 
evaluating trial function 'ft_trialfun_show'
reading the events from '/home/affneu/evefra/Spinalcord/data/Physio/subx017_SPINAL.vhdr'
the following events were found in the data
event type: 'New Segment' 
with event values: 

event type: 'Response' 
with event values: 'R  1' 

event type: 'Stimulus' 
with event values: 'S  1' 'S 97' 

no trials have been defined yet, see FT_DEFINETRIAL for further help
found 172 events
the call to "ft_definetrial" took 1 seconds and required the additional allocation of an estimated 11 MB
evaluating trial function 'ft_trialfun_general'
reading the header from '/home/affneu/evefra/Spinalcord/data/Physio/subx017_SPINAL.vhdr'
reading the events from '/home/affneu/evefra/Spinalcord/data/Physio/subx017_SPINAL.vhdr'
05-Jun-2023 12:40:54 - Failed  'TAPAS PhysIO Toolbox'
Error using ft_definetrial
no trials were defined, see FT_DEFINETRIAL for help

The following modules did not run:
Failed: TAPAS PhysIO Toolbox`

mrikasper, 2023-07-27:
Dear Evelyne,

My apologies for the delay in responding, conference season and now summer is upon us...

I presume this is BrainProducts data? We are relying on Fieltrip `ft_*` functions here and they might have changed since the implementation of that code. I am also no EEG analysis expert, so it would be great to get a better idea of the data.

Therefore, would you be able to share you SPM batch file and an example dataset, so that I can investigate further?

Thank you and all the best,
Lars

mrikasper, 2023-07-27:
And I just realized that the BrainProducts reader might not be functional at the moment, please see this issue: 

https://github.com/translationalneuromodeling/tapas/issues/225

As I stated there, it would be great to have some example data to implement unit and integration tests and make it work properly again. Please let me know if you are interested in pursuing this with me.