translationalneuromodeling/tapas #238: Calculating HFHRV and LFHRV

reemTamimi, 2023-07-27:
Hi, I am using the toolbox to calculate rmssd, hfhrv, and lfhrv (high and low frequency heart rate variability). I have a command which calculates the rmssd value (rms(diff(physio.ons_secs.cpulse)). Are there any similar commands that calculate the hfhrv and lfhrv values from the physio.mat file? 

Please let me know if you would like me to provide any additional information. Thank you!

mrikasper, 2023-07-27:
Dear @reemTamimi,

Would you have a frequency range for those low and high-frequency components or a literature reference for them. We are using low/high/bandpass filters in other parts of the toolbox, e.g., when preprocessing the cardiac and especially respiratory traces in these functions:

https://github.com/translationalneuromodeling/tapas/blob/604c56843c15411f5bd80190f81d845ac57d8592/PhysIO/code/preproc/tapas_physio_filter_respiratory.m#L127
https://github.com/translationalneuromodeling/tapas/blob/604c56843c15411f5bd80190f81d845ac57d8592/PhysIO/code/preproc/tapas_physio_filter_cardiac.m#L66

You could take this as a template for what you are trying to achieve. 

In general, maybe it is worthwhile thinking about the best way to extract derived measures from the processed physiological time series (cardiac/resp phase, RV(T), HR). This could be post-hoc processing functions to an already existing physio.mat.

All the best,
Lars

reemTamimi, 2023-08-07:
done

closed 2023-08-07 by reemTamimi