translationalneuromodeling/tapas #240: Error running example data

alexisganesh, 2023-08-09:
Hi! I think I've installed and followed along as outlined, but I can not get the example data to run. Different examples give different errors, which makes me think there's an an environment issue or I've done something wrong along the way (although it was an incredibly straightforward README file). Any help is appreciated!  

Here are some errors:

Biopac data:
09-Aug-2023 14:45:01 - Running 'TAPAS PhysIO Toolbox'
09-Aug-2023 14:45:01 - Failed  'TAPAS PhysIO Toolbox'
Error using fgets
Invalid file identifier. Use fopen to generate a valid file identifier.
In file "/media/lexy/f6f9acb0-022a-4a73-a08c-c30c0d343dc9/Neuroimaging/Softwares/tapas-master/PhysIO/code/readin/tapas_physio_read_columnar_textfiles.m" (???), function "tapas_physio_read_columnar_textfiles" at line 106.
In file "/media/lexy/f6f9acb0-022a-4a73-a08c-c30c0d343dc9/Neuroimaging/Softwares/tapas-master/PhysIO/code/readin/tapas_physio_read_physlogfiles_biopac_txt.m" (???), function "tapas_physio_read_physlogfiles_biopac_txt" at line 75.
In file "/media/lexy/f6f9acb0-022a-4a73-a08c-c30c0d343dc9/Neuroimaging/Softwares/tapas-master/PhysIO/code/readin/tapas_physio_read_physlogfiles.m" (???), function "tapas_physio_read_physlogfiles" at line 70.
In file "/media/lexy/f6f9acb0-022a-4a73-a08c-c30c0d343dc9/Neuroimaging/Softwares/tapas-master/PhysIO/code/tapas_physio_main_create_regressors.m" (???), function "tapas_physio_main_create_regressors" at line 137.
In file "/media/lexy/f6f9acb0-022a-4a73-a08c-c30c0d343dc9/Neuroimaging/Softwares/tapas-master/PhysIO/code/tapas_physio_cfg_matlabbatch.m" (???), function "run_physio" at line 1661.

The following modules did not run:
Failed: TAPAS PhysIO Toolbox

Philips data:
09-Aug-2023 14:52:40 - Running job #1
------------------------------------------------------------------------
09-Aug-2023 14:52:40 - Running 'TAPAS PhysIO Toolbox'
09-Aug-2023 14:52:40 - Failed  'TAPAS PhysIO Toolbox'
Error using tapas_physio_read_physlogfiles_philips_matrix
Physlog file not found: /home/lexy/SCANPHYSLOG.log
In file "/media/lexy/f6f9acb0-022a-4a73-a08c-c30c0d343dc9/Neuroimaging/Softwares/tapas-master/PhysIO/code/readin/tapas_physio_read_physlogfiles_philips_matrix.m" (???), function "tapas_physio_read_physlogfiles_philips_matrix" at line 41.
In file "/media/lexy/f6f9acb0-022a-4a73-a08c-c30c0d343dc9/Neuroimaging/Softwares/tapas-master/PhysIO/code/readin/tapas_physio_read_physlogfiles_philips.m" (???), function "tapas_physio_read_physlogfiles_philips" at line 67.
In file "/media/lexy/f6f9acb0-022a-4a73-a08c-c30c0d343dc9/Neuroimaging/Softwares/tapas-master/PhysIO/code/readin/tapas_physio_read_physlogfiles.m" (???), function "tapas_physio_read_physlogfiles" at line 84.
In file "/media/lexy/f6f9acb0-022a-4a73-a08c-c30c0d343dc9/Neuroimaging/Softwares/tapas-master/PhysIO/code/tapas_physio_main_create_regressors.m" (???), function "tapas_physio_main_create_regressors" at line 137.
In file "/media/lexy/f6f9acb0-022a-4a73-a08c-c30c0d343dc9/Neuroimaging/Softwares/tapas-master/PhysIO/code/tapas_physio_cfg_matlabbatch.m" (???), function "run_physio" at line 1661.

The following modules did not run:
Failed: TAPAS PhysIO Toolbox

alexisganesh, 2023-08-15:
Just in case anyone has one of those can't think kind of days like I was having and sees this/needs to figure it out  -- the biopac example data and the biopac example matlab script has some mismatched capitals :) Adjust the /examples/6.0.1/Physio/BioPac_txt/PPU3T/BioPac_Physio_Data_Export.txt file to : Biopac_Physio_Data_Export.txt and all works great! Thanks for the best toolbox around!

mrikasper, 2023-08-16:
Thank you for pointing this out, @alexisganesh !

We are aware of this issue and will fix it in the next release (coming up within a month!). It slipped through our tests, because they worked fine on Windows, which is case insensitive.

Sorry for the inconvenience and thank you for fixing the bug yourself! I am happy you are still considering PhysIO as the best toolbox around ;-)

All the best,
Lars