translationalneuromodeling/tapas #244: Philips Physiolog

Ruwanmri, 2023-09-06:
Hi,

I am trying to do physiological noise correction on Philips fmri and I want to use Scan/Physiolog time Synchronisation with gradient_log for the grad direction of Z. Could you please advise how to correctly set up the batch editor for Philips SCANPHYSIOLOGS. I get an error 'Error using tapas_physio_log'. 

Thank you.

Best regards,
Ru

mrikasper, 2023-09-06:
Dear Ru,

Thank you for using PhysIO!

This is indeed the trickiest part when setting up Philips physiological logfile synchronization. Luckily, you only have to do it once per study, because the gradient time courses should only depend on sequence parameters (nSlices, TR, resolution, parallel imaging acceleration, angulation etc.), but are usually not subject-dependent.

A good starting point to understand the different parameters is [this question in the FAQ](https://gitlab.ethz.ch/physio/physio-doc/-/wikis/FAQ#12-philips-i-would-like-to-use-the-gradient-log-for-timing-synchronization-but-how-do-i-set-the-thresholds). Basically, you will have to find a gradient direction, whose regularity reflects the onsets of slice and/or volume acquisition best and then set the thresholds in a way that you null local maxima (`.zero` threshold parameter), but keep the slice or volume peaks in the thresholded gradient time series (`.slice` and `.vol` threshold, respectively), as it is depicted in the middle panel of the Gradient Tresholding figure.

Let me know if this helps!

All the best,
Lars

Ruwanmri, 2023-09-07:
Hi Lars,

Thank you for explaining this. I went through the parameters and figured out the values for our data. However, I use the onset slice as 1 because it does not work with the 27th slice, but the toolbox manual says better to use the middle slice. so what is the standard onset slice for this setup? I also get a warning like below, any idea what input is wrong there?

`Warning: First volume with missing slice events (2317: 5 instead of 54 found)`

Thank you very much for the help.

Best regards,
Ru

Ruwanmri, 2023-09-08:
Hi Lars,

Could you please check these values based on the figure I get with the default settings of the batch editor? Attached a figure, I still get errors with my setup. Could you please have a look?

`matlabbatch{1}.spm.tools.physio.save_dir = {'C:\Users\ppzrw2\Documents\transfer_da
[image]
ta\mTBI_CHBH_IHN_SPMIC\spmic_rawdata\physilog_raw\CRT_run1\physio'};
matlabbatch{1}.spm.tools.physio.log_files.vendor = 'Philips';
matlabbatch{1}.spm.tools.physio.log_files.cardiac = {'C:\Users\ppzrw2\Documents\transfer_data\mTBI_CHBH_IHN_SPMIC\spmic_rawdata\physilog_raw\CRT_run1\SCANPHYSLOG20230616130941.log'};
matlabbatch{1}.spm.tools.physio.log_files.respiration = {'C:\Users\ppzrw2\Documents\transfer_data\mTBI_CHBH_IHN_SPMIC\spmic_rawdata\physilog_raw\CRT_run1\SCANPHYSLOG20230616130941.log'};
matlabbatch{1}.spm.tools.physio.log_files.scan_timing = {''};
matlabbatch{1}.spm.tools.physio.log_files.sampling_interval = [];
matlabbatch{1}.spm.tools.physio.log_files.relative_start_acquisition = 0;
matlabbatch{1}.spm.tools.physio.log_files.align_scan = 'last';
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Nslices = 54;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.NslicesPerBeat = [];
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.TR = 2;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Ndummies = 2;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Nscans = 138;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.onset_slice = 1;
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.time_slice_to_slice = [];
matlabbatch{1}.spm.tools.physio.scan_timing.sqpar.Nprep = 1;
matlabbatch{1}.spm.tools.physio.scan_timing.sync.gradient_log.grad_direction = 'z';
matlabbatch{1}.spm.tools.physio.scan_timing.sync.gradient_log.zero = 0.67;
matlabbatch{1}.spm.tools.physio.scan_timing.sync.gradient_log.slice = 0.76;
matlabbatch{1}.spm.tools.physio.scan_timing.sync.gradient_log.vol = [];
matlabbatch{1}.spm.tools.physio.scan_timing.sync.gradient_log.vol_spacing = [];
matlabbatch{1}.spm.tools.physio.preproc.cardiac.modality = 'PPU_Wifi';
matlabbatch{1}.spm.tools.physio.preproc.cardiac.filter.no = struct([]);
matlabbatch{1}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.min = 0.4;
matlabbatch{1}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.file = 'initial_cpulse_kRpeakfile.mat';
matlabbatch{1}.spm.tools.physio.preproc.cardiac.initial_cpulse_select.auto_matched.max_heart_rate_bpm = 90;
matlabbatch{1}.spm.tools.physio.preproc.cardiac.posthoc_cpulse_select.off = struct([]);
matlabbatch{1}.spm.tools.physio.preproc.respiratory.filter.passband = [0.01 2];
matlabbatch{1}.spm.tools.physio.preproc.respiratory.despike = false;
matlabbatch{1}.spm.tools.physio.model.output_multiple_regressors = 'multiple_regressors.txt';
matlabbatch{1}.spm.tools.physio.model.output_physio = 'physio.mat';
matlabbatch{1}.spm.tools.physio.model.orthogonalise = 'none';
matlabbatch{1}.spm.tools.physio.model.censor_unreliable_recording_intervals = false;
matlabbatch{1}.spm.tools.physio.model.retroicor.yes.order.c = 3;
matlabbatch{1}.spm.tools.physio.model.retroicor.yes.order.r = 4;
matlabbatch{1}.spm.tools.physio.model.retroicor.yes.order.cr = 1;
matlabbatch{1}.spm.tools.physio.model.rvt.no = struct([]);
matlabbatch{1}.spm.tools.physio.model.hrv.no = struct([]);
matlabbatch{1}.spm.tools.physio.model.noise_rois.no = struct([]);
matlabbatch{1}.spm.tools.physio.model.movement.no = struct([]);
matlabbatch{1}.spm.tools.physio.model.other.no = struct([]);
matlabbatch{1}.spm.tools.physio.verbose.level = 2;
matlabbatch{1}.spm.tools.physio.verbose.fig_output_file = '';
matlabbatch{1}.spm.tools.physio.verbose.use_tabs = false;`

Thank you.
Ru

closed 2023-09-17 by Ruwanmri