translationalneuromodeling/tapas #246: gradient_choice error

Ruwanmri, 2023-09-17:
Hi, 

I get the following error for the physio.scan_timing.sync.method = 'gradient_log';  when I use physio matlab script without SPM.

`Unrecognized function or variable 'gradient_choice'.

Error in tapas_physio_create_scan_timing_from_gradients_philips (line 164)
gradient_choice         = reshape(gradient_choice, [] ,1);

Error in tapas_physio_create_scan_timing (line 83)
            tapas_physio_create_scan_timing_from_gradients_philips( ...

Error in tapas_physio_main_create_regressors (line 165)
            [ons_secs, VOLLOCS, LOCS, verbose] = tapas_physio_create_scan_timing(...

Error in nospm_philips (line 76)
physio = tapas_physio_main_create_regressors(physio);`

I use the script as follows;

`%% Example script using PhysIO with Matlab only (no SPM needed)
%  For documentation of the parameters, see also tapas_physio_new (e.g., via edit tapas_physio_new)

%% Create default parameter structure with all fields
physio = tapas_physio_new();

%% Individual Parameter settings. Modify to your need and remove default settings
physio.save_dir = {'C:\Users\ppzrw2\Documents\transfer_data\retroicor\philips\crt_run1\physio_nospm'};
physio.log_files.vendor = 'Philips';
physio.log_files.cardiac = {'C:\Users\ppzrw2\Documents\transfer_data\retroicor\philips\crt_run1\SCANPHYSLOG20230616130941.log'};
physio.log_files.respiration = {'C:\Users\ppzrw2\Documents\transfer_data\retroicor\philips\crt_run1\SCANPHYSLOG20230616130941.log'};
physio.log_files.scan_timing={''};
physio.log_files.sampling_interval=[];
physio.log_files.relative_start_acquisition = 0;
physio.log_files.align_scan = 'last';
physio.scan_timing.sqpar.Nslices = 54;
physio.scan_timing.sqpar.TR = 2;
physio.scan_timing.sqpar.Ndummies = 2;
physio.scan_timing.sqpar.Nscans = 138;
physio.scan_timing.sqpar.onset_slice = 27;
physio.scan_timing.sync.method = 'gradient_log';
physio.scan_timing.sync.gradient_log.grad_direction = 'z';
physio.scan_timing.sync.gradient_log.zero=0.66;
physio.scan_timing.sync.gradient_log.slice=0.74;
physio.scan_timing.sync.gradient_log.vol = [];
physio.scan_timing.sync.gradient_log.vol_spacing = 0.08;
physio.preproc.cardiac.modality = 'PPU';
physio.preproc.cardiac.filter.include = false;
physio.preproc.cardiac.filter.type = 'butter';
physio.preproc.cardiac.filter.passband = [0.3 9];
physio.preproc.cardiac.initial_cpulse_select.method = 'auto_matched';
physio.preproc.cardiac.initial_cpulse_select.max_heart_rate_bpm = 90;
physio.preproc.cardiac.initial_cpulse_select.file = 'initial_cpulse_kRpeakfile.mat';
physio.preproc.cardiac.initial_cpulse_select.min = 0.4;
physio.preproc.cardiac.posthoc_cpulse_select.method = 'off';
physio.preproc.cardiac.posthoc_cpulse_select.percentile = 80;
physio.preproc.cardiac.posthoc_cpulse_select.upper_thresh = 60;
physio.preproc.cardiac.posthoc_cpulse_select.lower_thresh = 60;
physio.preproc.respiratory.filter.passband = [0.01 2];
physio.preproc.respiratory.despike = false;
physio.model.orthogonalise = 'none';
physio.model.censor_unreliable_recording_intervals = false;
physio.model.output_multiple_regressors = 'multiple_regressors.txt';
physio.model.output_physio = 'physio.mat';
physio.model.retroicor.include = true;
physio.model.retroicor.order.c = 3;
physio.model.retroicor.order.r = 4;
physio.model.retroicor.order.cr = 1;
physio.model.rvt.include = false;
physio.model.rvt.method = 'hilbert';
physio.model.rvt.delays = 0;
physio.model.hrv.include = false;
physio.model.hrv.delays = 0;
physio.model.noise_rois.include = false;
physio.model.noise_rois.thresholds = 0.9;
physio.model.noise_rois.n_voxel_crop = 0;
physio.model.noise_rois.n_components = 1;
physio.model.noise_rois.force_coregister = 1;
physio.model.movement.include = false;
physio.model.movement.order = 6;
physio.model.movement.censoring_threshold = 0.5;
physio.model.movement.censoring_method = 'FD';
physio.model.other.include = false;
physio.verbose.level = 2;
physio.verbose.process_log = cell(0, 1);
physio.verbose.fig_handles = zeros(1, 0);
physio.verbose.use_tabs = false;
physio.verbose.show_figs = true;
physio.verbose.save_figs = false;
physio.verbose.close_figs = false;
physio.ons_secs.c_scaling = 1;
physio.ons_secs.r_scaling = 1;
physio.version = 'R2022a-v8.1.0';

%% Run physiological recording preprocessing and noise modeling
physio = tapas_physio_main_create_regressors(physio);`

could you please help me with the error here? any clue would be much appreciated. 

Thanks,
Ru

mrikasper, 2023-09-19:
Dear Ru,

This is weird, it seems as if the `sync.grad_direction` is not properly defined. Would you be able to go into `tapas_physio_create_scan_timing_from_gradients_philips.m`, set a breakpoint at line 164 and tell me what the output is if you type ``sync.grad_direction` in the command window, please?

All the best,
Lars

Ruwanmri, 2023-09-19:
Hi,

Thanks for this. Sorry, I did not get your point. Did you mean to type sync.grad.direction without running the script? I got the same error when I ran the nospm physio script with the debug point at 164 on tapas_physio_create_scan_timing_from_gradients_philips.m.

Please correct me if this is not right.

Thank you.
Ru

mrikasper, 2023-09-19:
Dear Ruwan,

If you open the function, you should be able to set a breakpoint (red dot) next to line 164. After that, rerun your script, and when it stops at the breakpoint, type in sync.grad_direction in the command window.

I hope that helps!

All the best,
Lars

Ruwanmri, 2023-09-19:
Hi Lars,

Thanks. yeah, that is what I did and I get the same error. It does not run up to 164, I guess. 

`Unrecognized function or variable 'gradient_choice'.

Error in tapas_physio_create_scan_timing_from_gradients_philips (line 164)
gradient_choice         = reshape(gradient_choice, [] ,1);

Error in tapas_physio_create_scan_timing (line 83)
            tapas_physio_create_scan_timing_from_gradients_philips( ...

Error in tapas_physio_main_create_regressors (line 165)
            [ons_secs, VOLLOCS, LOCS, verbose] = tapas_physio_create_scan_timing(...

Error in nospm_philips (line 80)
physio = tapas_physio_main_create_regressors(physio);`

Thanks,
Ru